whitequark/solvespace #119: Export 3d Wireframe firstly propose name "untitled.AutoCAD 2007"

Symbian9, 2016-04-17:
In latest nightly when try open export to wireframe dialog it firstly propose name of file as "untitled.AutoCAD 2007", but as look near Save button you will see that it must be STEP File.

When try select DXF directly, it propose normal "untitled.dxf" 

P.S.: Also, propose use in interface definitions "2D" and "3D" instead of "2d" and "3d" that with small letters is not correct.

whitequark, 2016-04-17:
Not a bug. You have stale settings from back when the extension was determined incorrectly. Simply select DXF or STEP, re-save and this won't happen again.

closed 2016-04-17 by whitequark

Symbian9, 2016-04-17:
@whitequark, Ok, but anyway, try use 2D and 3D instead of 2d and 3d in user interface.
Thanks.

whitequark, 2016-04-17:
Both spellings are acceptable, like "DC" and "dc" for direct current.