whitequark/solvespace #120: Allow use millimeters units in system line styles too

Symbian9, 2016-04-17:
For now, system styles use only pixels, but some styles still used for export, so, allow use millimeters for them too (now its possible only in newly created styles)

For example, hidden lines cant be assigned to other newly created style, but units locked to pixels only will give not good output if need follow drawing standarts.

whitequark, 2016-04-17:
That's not a solution because switching edge styles, etc, to millimeters makes rendered output ugly instead of exported output. In any case, the vector graphics export right now is wildly inconsistent with any existing standards; for now, export to DXF and then manually fix it in LibreCAD or any other DXF editor. The lines are exported into layers according to their style, so it is very easy to change their thickness all at once.

closed 2016-04-17 by whitequark

Symbian9, 2016-04-17:
So, maybe styles in future will have two dimension:
* in pixels - for rendering
* in millimetres - for vector output (used specially for export)