whitequark/solvespace #122: Const correctness

whitequark, 2016-04-17:
Add const qualifiers where applicable, and convert copies of nontrivial structures to const references.

Evil-Spirit, 2016-05-11:
Some work in progress.

whitequark, 2016-05-18:
This will need to be rebased on top of current master.

Evil-Spirit, 2016-05-18:
Ok, I am doing this at now.

Evil-Spirit, 2016-05-18:
Not rebased yet, but almost ready.

Evil-Spirit, 2016-05-19:
@whitequark, Finished and rebased.

Evil-Spirit, 2016-05-19:
I think it's need to integrate such big chages faster.

Evil-Spirit, 2016-05-20:
Insist on the integrations asap.

whitequark, 2016-05-20:
I'm working on it...

whitequark, 2016-05-20:
Well, there are problems.

1. constification of RgbaColor makes no sense. It is MORE expensive to pass it by reference than by value on 64-bit platforms.
2. `const T *` is outside of scope for this change and in many cases it makes no sense whatsoever (like where you pass it to GLUT, which then drops the const qualifier, or when you return it).

So I will now have to undo that.

whitequark, 2016-05-20:
> constification of RgbaColor makes no sense

Specifically transforming `RgbaColor` to `const RgbaColor &`, not changes to the class itself, those are fine.

Evil-Spirit, 2016-05-20:
1) oh, yes, this is just 
`uint8_t red, green, blue, alpha;`
2) Can you explain it using more examples, I don't understand what's wrong

Evil-Spirit, 2016-05-20:
> const T * is outside of scope
if you mean special cases - you are right, but in general you can not cancel const T *. In such places we must override it by const T&.

whitequark, 2016-05-20:
> Can you explain it using more examples, I don't understand what's wrong

This issue concerns one specific change: transforming arguments of type `T` to type `const T &`, so that we make less copies. This places restrictions on callee, which are mostly already satisfied.

In some places you have:

 * changed the type of return value from `T *` to `const T *`, instead, which makes no sense: it places restrictions on the *caller*. Maybe the caller *does* want to mutate the result.
 * changed the type of `fndata` to `const void *`, which makes no sense: the interface with C code in GLUT will cast that away and anyway, why shouldn't fndata be mutated? Again this places restrictions on the caller (which supplies the function).

whitequark, 2016-05-20:
Really we should have done this in two steps: step 1 adding const qualifiers to functions, step 2 changing signatures...

Evil-Spirit, 2016-05-20:
This canges are needed to support logical consts everywhere. You can not make const in some places. Once you introduce const, you have to make it everywhere where it logically.

Evil-Spirit, 2016-05-20:
Also, I never uses const_cast, so I have to made fndata const void *

whitequark, 2016-05-20:
No, they are not.

Evil-Spirit, 2016-05-20:
What are not?

whitequark, 2016-05-20:
When I revert a change and code compiles, it means it wasn't necessary after all.

Anyway, there's a larger problem with this. Specifically, adding `const` specifiers to functions is always safe, but adding `const T &` introduces aliasing, especially when we cannot guarantee that `T` cannot be mutated underneath us. So this is a cost of not copying. I wonder how to best handle this...

Evil-Spirit, 2016-05-20:
You mean use "const T &a" and make internal copy, or leave the "T a"?

whitequark, 2016-05-20:
Well, we can also make small classes like `Point2d` and such with immutable interior. I.e. `const double x, y;`. Then it is always safe to pass them by reference. But this is a drastic change and we need to measure the benefit before doing it, I think.

whitequark, 2016-05-20:
For now I will split your commits into two, one introducing `const` qualifier on functions and another adding `const T &`. This can be done with no manual work, just some regexes on patches.

whitequark, 2016-05-20:
Making internal copy is stupid, it provides exactly no benefit, and actually makes compiler optimizations work worse, so the only choice here is accept a `const T &` or keep `T`.

whitequark, 2016-05-21:
Also I just noticed you used `mutable`. Don't. It's even worse than const_cast.

whitequark, 2016-05-21:
In drawconstraint.cpp you wrote this:
```
static void LineCallback(const void *fndata, Vector a, Vector b)
{
    Constraint *c = (Constraint *)fndata;
    c->LineDrawOrGetDistance(a, b);
}
```

So you didn't use `const_cast`, you used a C-style cast, which is a superset of that. Which is exactly what I was talking about: you lied using `const`.

whitequark, 2016-05-21:
Also, there is absolutely no difference between accessing member through `this` or not.

Evil-Spirit, 2016-05-21:
> Also I just noticed you used mutable. Don't. It's even worse than const_cast.
You can not avoid mutable while in dogd. Drawing operation must be const anyway. Dogd is for temporary there. So, for now it's not the problem.

> Constraint *c = (Constraint *)fndata;
must be "const Constraint", ofc.

whitequark, 2016-05-21:
No, it is a problem. Using `mutable` makes `const` completely pointless. What should be done is extraction of caches from the data structures so that they themselves do not have to be mutated.

In case of dogd, the entire logic of dogd will be extracted in #125.

whitequark, 2016-05-21:
Anyway, I implemented a much more conservative version of this in 4fde94b. We'll have to decide when to actually use it later; I think the right way is to make geometry classes such as `Vector` use const fields and then just use references everywhere, but I'll need to think about it and see how troubling is that.

closed 2016-05-21 by whitequark

Evil-Spirit, 2016-05-21:
> No, it is a problem.
What problem?

whitequark, 2016-05-21:
`mutable`.

Evil-Spirit, 2016-05-21:
Just leave it for time when we kill dogd.

whitequark, 2016-05-21:
As you like to say, there's nothing more permanent than temporary hacks. So I will not.

Evil-Spirit, 2016-05-21:
We have not be so strict while the refactoring process. We have to introduce const NOW. There is no point to postprone it.

Evil-Spirit, 2016-05-21:
We must create ticket for mutable and will never forget it.

Evil-Spirit, 2016-05-21:
We can't use const inside "trivially possible", we must use it everywhere since we write const once.

whitequark, 2016-05-21:
That's not really true. Adding `const` is not an all-or-nothing deal, but rather it pulls in large islands of the code. For example all of our geometry is already covered. I think that the only two parts that are currently covered incompletely are `srf/` stuff and dogd, and `srf/` cannot be covered further without logic changes.

whitequark, 2016-05-21:
Most importantly neither of those can be fixed without logic changes, and with this commit I covered everything that does not involve logic changes. That's important. There can be basically no bugs in this commit. When we next remove dogd, etc, it will be far easier to review and if necessary fix it than if everything is lumped together.

Evil-Spirit, 2016-05-21:
mutable is not logic changes.

whitequark, 2016-05-21:
`mutable` isn't, and I am opposed to any `mutable`. Even if we remove `dogd` soon then the one in `srf` still remains.

Evil-Spirit, 2016-05-21:
Ok, so do it you own way. I just will write const or mutabe when I will get some trouble with it.

whitequark, 2016-05-21:
If we ever use `mutable` then let's just not use `const` instead because it's completely and utterly pointless. `const` is weak enough as it is in C++ to ignore it with `mutable`.