whitequark/solvespace #123: Enum-ification

whitequark, 2016-04-17:
Give names to enums and convert all `int` parameters and fields that are actually enums to use those names.

Use the `unreachable` intrinsic and Clang warnings to make sure that if/else chains and switches always cover all enum variants.

Evil-Spirit, 2016-05-20:
I think we should use
`enum Operation : uint8_t {`
to ensure the type size.

Evil-Spirit, 2016-05-20:
@whitequark, Just pushed draft. I need you opinion about:
* types of enums(always uint32_t or only for saved things)
* naming
* location (inside class, outside class, using "enum class", etc)

whitequark, 2016-05-20:
> always uint32_t or only for saved things

Yes, always uint32_t for now. This makes it easier to do bindings, because the type of enum is not specified in C or C++, and setting everything to just uint32_t removes that ambiguity. Also please don't change the numbering in this changeset.

> location (inside class, outside class, using "enum class", etc)

Just enum inside class everywhere. That should be fine in all cases I can see.

> naming

In e.g. `Group::GroupType` the second `Group` is extraneous, just call it `Group::Type`. In e.g. `ClassifyType` and `CombineType` it's better to call them `ClassifyHow` and `CombineHow`.

Evil-Spirit, 2016-05-20:
I think ClassifyHow not sounds like typename, but like variable name
```
void f(ClassifyHow how)  //  extra how
void f(Classify how)          // like real language, but in general I think it's bad practice
void f(ClassifyType how)  // sounds nomal
```

whitequark, 2016-05-20:
> void f(ClassifyHow how)   //  extra how

This is the same thing as "Group group" etc. The variable name is the shortening of type name. So I think that's OK.

Evil-Spirit, 2016-05-20:
> Just enum inside class everywhere
Note what we can't do it everywhere because we have poor header/cpp splitting, so in some number of cases we have to use enum before class definition.
I don't know, is where something like: 
```
// fwd decl
enum Class::Type : uint32_t;
```

whitequark, 2016-05-20:
Yes you can do that in C++11.

Evil-Spirit, 2016-05-20:
> Yes you can do that in C++11.
That's good.

whitequark, 2016-05-25:
Fixed in f33ddc94, c890d663c and e2ea8c027.

closed 2016-05-25 by whitequark