whitequark/solvespace #124: Boolean argument names

whitequark, 2016-04-17:
At all call sites, when a `true` or `false` value is passed to a `bool argName`, the call site should be amended to say `/*argName=*/true`.

ruevs, 2016-04-17:
Just for readability and to reduce chances of passing wrong values to arguments?

whitequark, 2016-04-17:
Yes; this deficiency, caused by the lack of keyword arguments in C++, tends to cause annoying, hard to find bugs, and I have fixed a fair share of them already.

I prefer eliminating bug classes rather than fixing bugs. An even stricter solution (that I ordinarily prefer) would be to get rid of booleans entirely and use enums instead (i.e. `bool allCoplanar` → `enum { ALL_COPLANAR, NOT_ALL_COPLANAR }` etc), but I fear this will be too unorthodox for SolveSpace.

ruevs, 2016-04-17:
There is a hidden risk in your proposal. If the interface is changed (e.g. arguments added, removed, renamed, or /why?!/ rearranged) then the "labeling" comments at all call sites have to be updated. If the person doing it is in a hurry, negligent, does not care or does not understand the convention there is a chance that they become stale. Later on someone will be wondering what was really intended.

whitequark, 2016-04-17:
Such a person should abstain from contributing, especially to a project written in C++, where excessive carelessness trivially leads to data loss and code execution vulnerabilities.

whitequark, 2016-04-17:
More on topic though--this is exactly why I prefer enums. They also handle the case where non-constants are passed. Unfortunately, this technique is extremely unconventional in C++...

ruevs, 2016-04-17:
"Like" on the first statement :-)
Forbidding "bool" is not so unconventional I have seen the practice in many embedded projects.

jwesthues, 2016-04-19:
For what it's worth, I often write bools for expediency, but would almost never object to their replacement with an enum in stable code, except in cases where the function name obviously implies the argument meaning.

whitequark, 2016-04-20:
Noted. I think there's a few more cases where (nominal, as opposed to structural) C++ enums are too heavyweight, `notAllCoplanar` and friends come to mind.

Evil-Spirit, 2016-05-25:
Pushed.

whitequark, 2016-05-26:
Fixed in 33bfad9. @Evil-Spirit Note that I omitted the `Printf` changes (too much churn, too little usefulness) and `/*x=*/x` (already obvious what it does, also wasn't in the issue description).

closed 2016-05-26 by whitequark

Evil-Spirit, 2016-05-26:
OK.