whitequark/solvespace #125: Implement offscreen rendering that is independent of GL implementation

whitequark, 2016-04-17:
This is needed to render reference images for regression testing.

Specifically:
  * Abstract the "draw GL", "add edges" and "get distance" codepaths behind a new class `Canvas`;
  * Add a `Canvas` implementation using [Cairo](http://www.cairographics.org/);
  * Use Cairo for PDF/SVG/etc rendering too? Maybe not.

whitequark, 2016-05-21:
@Evil-Spirit let's discuss this issue on IRC

whitequark, 2016-07-24:
@Evil-Spirit For this (for DrawPolygon) we need a tesselator that's independent of GL, i.e. not GLU. Do you have any idea which?

Evil-Spirit, 2016-07-24:
@whitequark, we can use ear-triangulation approach from triangulate.cpp

Evil-Spirit, 2016-07-24:
@whitequark, i am going to implement it soon

whitequark, 2016-07-24:
What are you implementing exactly?

Evil-Spirit, 2016-07-24:
Quads. Polygons will be the same just with triangulation.

Evil-Spirit, 2016-07-25:
@whitequark, polygons with native solvespace triangulation working good.

whitequark, 2016-07-27:
@Evil-Spirit So, how do I turn a polygon into triangles? Can you write some sample code? I've no idea how to use that part.

Evil-Spirit, 2016-07-27:
@whitequark, look here https://github.com/Evil-Spirit/solvespace/tree/canvas-opengl2

Evil-Spirit, 2016-07-27:
https://github.com/Evil-Spirit/solvespace/blob/canvas-opengl2/src/rendergl2.cpp#L1607

Evil-Spirit, 2016-07-27:
new function here https://github.com/Evil-Spirit/solvespace/blob/canvas-opengl2/src/srf/triangulate.cpp#L421

whitequark, 2016-08-01:
Implemented.

closed 2016-08-01 by whitequark