whitequark/solvespace #132: Store constraint label offset in local csys.

Evil-Spirit, 2016-04-18:
This will bring good behaviour when we transform sketch - the dimensions have not to jump unwanted way.

Evil-Spirit, 2016-04-18:
[image]

Evil-Spirit, 2016-04-18:
[image]

Evil-Spirit, 2016-04-18:
[image]

whitequark, 2016-04-18:
Transform?

Evil-Spirit, 2016-04-18:
For ex. change dimension value.

whitequark, 2016-04-18:
Ah yes, this sounds nice. But we need to do something with existing files.

Evil-Spirit, 2016-04-18:
No problem if we load old, save new approach. If we need, we can save pos in both csys.

whitequark, 2016-04-18:
That will wait until we have regression tests for savefile format, then.

whitequark, 2016-11-27:
Migrated to https://github.com/solvespace/solvespace/issues/122.

closed 2016-11-27 by whitequark