whitequark/solvespace #134: Certain extrusions broken sometime after 2.0

whitequark, 2016-04-19:
See comment by JujuSL in http://solvespace.com/forum.pl?action=viewthread&parent=1156 from 2016-04-18/19.

whitequark, 2016-04-23:
Root cause is the commit 09f59ddbc04. @Evil-Spirit, can you take a look?

Evil-Spirit, 2016-04-23:
Ok

whitequark, 2016-04-23:
Oh, -fsanitize=undefined already found the missing NULL.

whitequark, 2016-04-23:
`/home/whitequark/Work/solvespace/src/srf/boolean.cpp:806:11: runtime error: null pointer passed as argument 1, which is declared to never be null`

whitequark, 2016-04-23:
Ah no, that's an unrelated problem; it doesn't like a qsort(NULL, 0) invocation.

whitequark, 2016-05-04:
@Evil-Spirit Now opening that file crashes:
```
981                         if(tr->meta.face != info.tr->meta.face && info.count == 1) {
(gdb) bt
#0  0x000000000053ef42 in SolveSpace::SKdNode::MakeCertainEdgesInto (this=0x1f60790, sel=0x17a1880, how=400, coplanarIsInter=false, inter=0x0, leaky=0x0, 
    auxA=0) at /home/whitequark/Work/solvespace/src/mesh.cpp:981
#1  0x000000000053f6a3 in SolveSpace::SMesh::MakeCertainEdgesAndOutlinesInto (this=0x17a1848, sel=0x17a1880, sol=0x17a1890, type=400)
    at /home/whitequark/Work/solvespace/src/mesh.cpp:96
#2  0x000000000053bada in SolveSpace::Group::GenerateDisplayItems (this=0x17a1690) at /home/whitequark/Work/solvespace/src/groupmesh.cpp:411
#3  0x000000000052fff2 in SolveSpace::GraphicsWindow::LoopOverPoints (this=0xbd92f0 <SolveSpace::SS+16>, 
    entities=std::vector of length 3, capacity 4 = {...}, faces=std::vector of length 0, capacity 0, pmax=0x7fffffffc1a0, pmin=0x7fffffffc190, 
    wmin=0x7fffffffc188, usePerspective=false, includeMesh=true) at /home/whitequark/Work/solvespace/src/graphicswin.cpp:351
#4  0x000000000053063c in SolveSpace::GraphicsWindow::ZoomToFit (this=0xbd92f0 <SolveSpace::SS+16>, includingInvisibles=false, useSelection=false)
    at /home/whitequark/Work/solvespace/src/graphicswin.cpp:407
#5  0x000000000055dda9 in SolveSpace::SolveSpaceUI::AfterNewFile (this=0xbd92e0 <SolveSpace::SS>) at /home/whitequark/Work/solvespace/src/solvespace.cpp:341
#6  0x000000000055dfcc in SolveSpace::SolveSpaceUI::OpenFile (this=0xbd92e0 <SolveSpace::SS>, filename="/home/whitequark/Downloads/support4bis.slvs")
    at /home/whitequark/Work/solvespace/src/solvespace.cpp:141
#7  0x00000000005609bc in SolveSpace::SolveSpaceUI::MenuFile (id=101) at /home/whitequark/Work/solvespace/src/solvespace.cpp:461
```

Evil-Spirit, 2016-05-04:
Sorry, that was fixed at https://github.com/Evil-Spirit/solvespace/commit/18a37651d9c4f2973ba83297ee16a8faa7348cb6

whitequark, 2016-05-04:
Fixed in f5d8b1dc and a75bf6e2.

closed 2016-05-04 by whitequark