whitequark/solvespace #139: Replace `if(cond) oops();` with `assert(!cond && "Explanation);`

whitequark, 2016-04-22:
Right now, SolveSpace's oopses are quite unhelpful--there are many relatively common mistakes that trigger certain oopses but you have to go back to the source and sometimes ponder a bit as to what exactly the condition checks. I propose using assertions following the same pattern as LLVM does (see title).

LLVM's assertions look something like this (without file/line for clarity):
  * `llvm::MachineBasicBlock* llvm::MachineOperand::getMBB() const: Assertion `isMBB() && "Wrong MachineOperand accessor"`
  * `typename llvm::cast_retty<X, Y*>::ret_type llvm::cast(Y*) [with X = llvm::PointerType; Y = const llvm::Type; typename llvm::cast_retty<X, Y*>::ret_type = const llvm::PointerType*]: Assertion `isa<X>(Val) && "cast<Ty>() argument of incompatible type!"' failed.`

Most of the time it is not even necessary to know what the filename is as the logic error is self-apparent.

whitequark, 2016-04-22:
@jwesthues Any opinions on this? This is something I propose for new code, not a mass replacement, mostly because sketch consistency checking has to be reworked anyway before exposing it to user-written scripting.

Regarding the issue with `abort()` on Windows and the scary message, we can turn that off globally in debug builds using [_set_abort_behavior](https://msdn.microsoft.com/en-us/library/e631wekh.aspx).

Regarding release builds, I think it makes sense to override the CMake defaults and enable assertions in them as well.

jwesthues, 2016-04-22:
It's kind of ugly to have a mixture of (admittedly, nonstandard) oops() and assert(x). Implement oops("message")? Though I typically set oops() just to enter the debugger, at which point the context becomes relatively obvious.

whitequark, 2016-04-22:
The intent is to eventually get rid of all `oops()`, converting most of them to `assert(x)` and the rest to some sort of error propagation, probably exceptions.

whitequark, 2016-04-22:
(Re: exceptions, I'm well aware of their issues. I have a specific usage pattern in mind that I think allows recovering from broken invariants in sketch inputs without being too costly or obtuse, I'll elaborate on that later.)

jwesthues, 2016-04-22:
An `oops()` should generally be a "can't happen" condition of interest only to developers, and it's often in the `default:` clause of a `switch` statement or the last `else` clause, which doesn't transform nicely to `assert(condition)`. I see how e.g. scripting could change some of that, of course.

whitequark, 2016-04-22:
Indeed. I should mention that for such a `default` or `else` clause I would like to define a separate `unreachable()` function that accepts an optional message.

I do not dispute the nature of the `oops()`--`assert()` also specifies "can't happen conditions" for that matter. The main benefits of this change are 1) explicitly specifying the intent, because the context is often unclear: what does the condition in `if(data == NULL) oops();` mean? whereas `assert(data == NULL && "Failed to load data from file");` is self-describing, and 2) shortening the debug cycles in the case where an assertion is violated.

I don't know what sort of tooling do you use, but on *nix there are basically no low-friction debuggers, and having to drop into one is always an annoyance. LLVM even traps `SIGABRT` to print a backtrace when an assertion is failed, so that one would have to run the debugger as rarely as possible, which we may consider doing as well (it's a one-liner on glibc Linux systems).

whitequark, 2016-04-22:
I guess one possibility would be to convert `oops()` to `unreachable()` and afterwards, or simultaneously, add more descriptive messages to the cases where a condition is used.

jwesthues, 2016-04-22:
My typical setup automatically enters WinDbg upon `oops()`, with that line highlighted in the source, and all the usual debugger tools (call stack, inspect variables, etc.) available. I can see how the current behavior would be unhelpful without that.

Is there really no good way to achieve that under Linux? I've never tried, but that seems like a significant omission.

whitequark, 2016-04-22:
IDEs with debugger integration exist, although the good one (JetBrains Clion) is not only paid but also is subscription-only aka ships with a hostile DRM. The majority of people do not use the IDEs; I use Sublime Text, which is an excellent editor but has no debugger integration. I expect most other *nix contributors to use Vim, which is in a similar situation. In practice, most people use gdb with its baroque interface to debug C/C++ code under *nix.

Regardless, there is no concept of just-in-time debugging on *nix, i.e. one has to explicitly run the process under the debugger in the first place to be able to debug a crash in it, and depending on the frontend (e.g. Qt Creator, which is the best free IDE, suffers from this) it may take a few seconds for the frontend to launch the program under the debugger. I currently have to restart the program manually under gdb every time I want to debug an oops.

jwesthues, 2016-04-22:
I don't use any IDE, just the debugger. I write code with just a text editor.

gdb certainly can attach to a running process. Maybe oops() should just freeze under *nix, and wait forever to let you do that?

whitequark, 2016-04-22:
That seems quite hostile to people who do not intend to debug crashes.

whitequark, 2016-04-22:
Speaking of people who do not intend to debug crashes--I have users filing issues with `stderr` output attached, as is common, and this would make it more useful as well.

jwesthues, 2016-04-22:
I'd hope that non-developer users aren't encountering an `oops()` so frequently as to make their experience in that case a big concern. But isn't the file/line already printed to stderr upon `oops()` now?

whitequark, 2016-04-23:
This mostly matters for nightlies, and getting feedback on those is important... at least until we have good test coverage. The file and line are indeed printed, though that might be misleading if the source has changed since. It's a minor convenience in this case.

jwesthues, 2016-04-23:
Fair point. Put the top commit hash in there or something? In any case, no objection to adding explanation to the internal errors; the mixture of conditional (`assert(!cond)`) and unconditional (`oops()`) internal errors just seemed ugly to me, and the `!cond && "Explanation"` trick to get the explanation string in, while somewhat common, is not too elegant.

whitequark, 2016-04-23:
OK. An alternative to that trick would be defining our own macro (and their respective functions), `ssassert` perhaps, that accept a message explicitly. I have no preference on this; I'll do it that way if you want.

jwesthues, 2016-04-23:
That seems cleaner to me, at least. I'd probably define a macro that accepts a message but no condition; but if really want a conditional internal error, then I'd rather do it consistently (and replace all the `oops()` with `ssassert(0, "message")` than do a mix.

whitequark, 2016-04-23:
OK. An explicit `ssunreachable("message")` perhaps?

jwesthues, 2016-04-23:
Why "unreachable"? That makes me think of code that a compiler could prove will never execute, not run-time conditions that aren't supposed to happen. If `oops()` is too casual, then maybe something more like `fail()`, or just the `ssassert(0, ...)` is a pretty common idiom.

whitequark, 2016-04-23:
As far as I'm aware it's a common term for the kind of assertion that should never execute. But I have no strong prefernce here, `ssassert(0, ...)` is fine.

jwesthues, 2016-04-23:
That gets it down to one function, at least, seems relatively elegant. I also don't object to printing other stuff post-assert, like a stack trace or whatever, if that would help debug without a debugger.

whitequark, 2016-05-19:
Implemented in 8e69df7040cf and fc63929526.

This is how assertion failures look like on Linux and OS X now:
```
File /home/whitequark/Work/solvespace/src/platform/unixutil.cpp, line 55, function ssfopen:
Assertion 'Unexpected null byte in middle of a path' failed: ((filename.length() != strlen(filename.c_str())) == false).
Backtrace:
 0: ./build/src/solvespace(_ZN10SolveSpace14assert_failureEPKcjS1_S1_S1_+0x7d) [0x548d2d]
 1: ./build/src/solvespace(_ZN10SolveSpace7ssfopenERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEPKc+0x7c) [0x548eac]
 2: ./build/src/solvespace(_ZN10SolveSpace12LoadResourceERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEPm+0x38e) [0x54d47e]
 3: ./build/src/solvespace(_ZN10SolveSpace7LoadPNGERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x2a) [0x60933a]
 4: ./build/src/solvespace(main+0x1e0) [0x54d900]
 5: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7fcc24990610]
 6: ./build/src/solvespace(_start+0x29) [0x52a199]
Aborted
```

@Evil-Spirit can you review fc63929526? Maybe messages can be improved, or I screwed up a condition somewhere...

Evil-Spirit, 2016-05-19:
Ok, will look it

Evil-Spirit, 2016-05-20:
I have issues to run this build.

Evil-Spirit, 2016-05-20:
Also, can we insert msg box inside ssassert to have ability to make bugreports by users? Also, we can make it something like "The bug has occured(file:line,msg etc). Do you want to try save the last undo state to file?".

whitequark, 2016-05-20:
> I have issues to run this build.

Well, are you going to describe which?

whitequark, 2016-05-20:
> Also, can we insert msg box inside ssassert to have ability to make bugreports by users?

Better ensure users never trigger assertion failures.

> Also, we can make it something like "The bug has occured(file:line,msg etc). Do you want to try save the last undo state to file?".

Definitely not. The autosave function is already there, and once an internal invariant is violated, we cannot guarantee *anything* about the model.

Evil-Spirit, 2016-05-20:
I don't figured out yet why exactly, but this is just crash on startup.

Evil-Spirit, 2016-05-20:
> Better ensure users never trigger assertion failures.

How can we do it without users bugreports? Then for what you wrote all this messages if it never occurs?

> Definitely not. The autosave function is already there, and once an internal invariant is violated, we cannot guarantee anything about the model.

Not guarantee, but attempt.

whitequark, 2016-05-20:
> How can we do it without users bugreports?

With our exhaustive internal testing.

> Then for what you wrote all this messages if it never occurs?

To make it easier to debug failures during our internal testing.

> Not guarantee, but attempt.

And maybe save broken files? No. We should *never* save files that can possibly be broken.

Evil-Spirit, 2016-05-20:
> And maybe save broken files? No. We should never save files that can possibly be broken

I don't mean over the original, just another name...
Anyway, it's not so important while we have autosave.

whitequark, 2016-05-20:
In the future I want sketch generation to happen on a background thread, which will inherently require a copy of the sketch. Then we could do this safely.

whitequark, 2016-05-20:
Fixed the bug, it was one inverted condition.

closed 2016-05-20 by whitequark