whitequark/solvespace #141: Use uint64_t for handle id value.

Evil-Spirit, 2016-04-30:
When we are importing big files, there is situation when handle limit reached.
In current implementation we can haldle only 64k of handles(half bits of uint32_t).

whitequark, 2016-05-04:
This seems somewhat excessive. I think we are going to fall over in other places (rendering, solver, etc) before we bump into this limit.

Evil-Spirit, 2016-05-04:
no problem to render and solve only 64k of lines (especially for only H/V/PC like in architectural drawings)

Evil-Spirit, 2016-05-04:
Of course, in the future, with small fixes of the solver. But since this will be problem if we make new file formats without considering this.

whitequark, 2016-05-04:
Makes sense.

Evil-Spirit, 2016-08-13:
inside this improvement we can review the handle generation logic to avoid bit-combine operations and remapping. @jwesthues, any ideas how to avoid remap and don't save the remap lists?

whitequark, 2016-08-13:
> to avoid bit-combine operations and remapping

Why are you trying to avoid them?

Evil-Spirit, 2016-08-13:
@whitequark, remapping is something what we have to save into file. We could think and invent new rules for id generation without remapping. Also, we have to introduce 64bit ids and we just can use union representation with 8/16/32 bit fields to make code simplier and cleaner.

whitequark, 2016-10-14:
Migrated to solvespace/solvespace

closed 2016-10-14 by whitequark