whitequark/solvespace #142: Introduce antialiasing for fat lines.

Evil-Spirit, 2016-04-30:
There is only for lines with width <= 3.0.

whitequark, 2016-05-04:
Before fat-line-aa:
[image]

After fat-line-aa:
[image]

I don't see any difference.

Evil-Spirit, 2016-05-04:
Which card/os? Do you have AA on the regular lines?

whitequark, 2016-05-04:
Intel, Linux

These are regular lines (closed shape to remove red background, but it's AA even with open shape):
[image]

Evil-Spirit, 2016-05-04:
Which Intel and which linux?

whitequark, 2016-05-04:
This should be enough: [image]

The release is Debian testing.

Evil-Spirit, 2016-05-04:
[image]

whitequark, 2016-05-04:
Well, yes. Gallium is the Mesa software renderer. It implements features of GL_LINES that no one else bothers to

Evil-Spirit, 2016-05-04:
There is the same lines used for aa, just need to be adjusted.

whitequark, 2016-05-04:
How about just postponing this until we have GLES rendering? GL_LINES don't work reliably and we should stop using them completely instead.

Evil-Spirit, 2016-05-04:
OK

closed 2016-05-04 by whitequark