whitequark/solvespace #143: Change behaviour of button "Hide shaded view of solid model".

Evil-Spirit, 2016-04-30:
In previous versions and now, this button is not only hides shaded mesh, but also shows "hidden" invisible lines, which is not stippled, but just like "wireframe".
I am offering to change this behaviour because now it is impossible to export view to pdf/eps/svg without faces, but with stippled hidden lines (or just edges with hidden lines removed).

Evil-Spirit, 2016-05-04:
Also, hiding edges causes hiding of outlines, but this is an interesting appearance mode (like in toon-shading).

Evil-Spirit, 2016-05-17:
Can we include this in 2.1?

whitequark, 2016-05-17:
I'll look into it

whitequark, 2016-06-01:
> I am offering to change this behaviour because now it is impossible to export view to pdf/eps/svg without faces, but with stippled hidden lines (or just edges with hidden lines removed).

But you can do that. Turn off view of shaded model, turn on hidden lines.

whitequark, 2016-06-01:
Just exported this
[image]

Evil-Spirit, 2016-06-02:
Turn off hidden, this mode is not supported.

whitequark, 2016-06-11:
> Turn off hidden, this mode is not supported.

Yeah, we can't do it for either viewport or export. I don't think it's important enough to get into 2.1, so denominating for backport.

whitequark, 2016-08-13:
@jwesthues Any idea for the UI for this feature? I'm not sure what to do with the proliferation of buttons. Maybe a tri-state one for hidden lines? Shown, stippled, hidden?

jwesthues, 2016-08-13:
Tri-state seems reasonable to me.

whitequark, 2016-08-13:
Tri-state button implemented in master.

closed 2016-08-13 by whitequark