whitequark/solvespace #144: Make showcase examples for site.

Evil-Spirit, 2016-05-03:


whitequark, 2016-05-03:
There's already solvespace.com...

closed 2016-05-03 by whitequark

Evil-Spirit, 2016-05-03:
For m-labs.hk page?

Evil-Spirit, 2016-05-03:
[image]

whitequark, 2016-05-03:
That's internal to m-labs, let's keep it off this issue tracker.

Evil-Spirit, 2016-05-03:
I've just wanted to make place where I can place it.

Evil-Spirit, 2016-05-03:
[image]

Evil-Spirit, 2016-05-03:
I ve already have some files made by me, so, why not to show it, may be it will be interesting.

whitequark, 2016-05-03:
Sure, you could make a repository, put the files inside and add screenshots to README.md, this way it will be all kept together.

Evil-Spirit, 2016-05-03:
Too complicated for mess of the files. The good ones we must publish on sites, small amount of the users care about github reops.

whitequark, 2016-05-03:
That is less messy than manually packaging stuff in zip archives, and a repository with a README.md in the root is as good as a website

Evil-Spirit, 2016-05-03:
We already have solvespace-testcases, so, we can do it there.

Evil-Spirit, 2016-05-03:
[image]