whitequark/solvespace #146: Don't consider arc center-point when calculating tolerances.

Evil-Spirit, 2016-05-04:
This causes problems on the arcs with big radius.

whitequark, 2016-05-04:
Can you show an example?

Evil-Spirit, 2016-05-04:
[image]

whitequark, 2016-05-04:
Ack.

Evil-Spirit, 2016-05-06:
Just leave it here for future:
[image]
[image]

Evil-Spirit, 2016-10-31:
fixed in https://github.com/Evil-Spirit/solvespace/commit/82c35b9809c88bdd66ac3c783320d4d54cbf712b
also, https://github.com/Evil-Spirit/solvespace/commit/4f7bc1ed9d4e10c6610aa060e4b1548ac26b2d5d

whitequark, 2016-11-02:
Merged.

closed 2016-11-02 by whitequark