whitequark/solvespace #147: Optimize GenerateALL for not doing unnecessary work.

Evil-Spirit, 2016-05-06:
When we doing some groups
1. sketch
2. extrude
3. sketch
4. extrude
5. sketch
...etc

We have decreased performace when editing the last sketch. The solve runs and mesh is generated only for the last group, but the entities and parameters are regenerated for the all groups, which is not acceptable.

The next things i see when it really works:
1. When we add new request which is generate new entities (but look at https://github.com/Evil-Spirit/solvespace/commit/f6ac2e002b1aed1a38435e32a56504ab140f369a);
2. When we add new group which is generate its entities;
3. When we delete request;
4. When we delete group;
5. When we load file and must regenereate all.

So, we don't need to regenetate all while we dragging some point, so we can introduce generation mode when we don't want to regen entities(because just parameters values changed only).
@jwesthues,
What do you think?

whitequark, 2016-05-06:
Ideally, I think a sketch would compute a dependency graph and only regenerate the minimal set.

Evil-Spirit, 2016-05-06:
Yes, but this needs to rework all the generation algorithm. But this is small thing I talking about is most annoying. It is problem just for ~16 groups.

Evil-Spirit, 2016-05-06:
We just move the point, but generation has to:

1. remove all the entities
2. store all the parameters in other place (just remember pointer)
3. recreate all the entities and parameters
4. copy old param values to new params (using handle lookup with o(log(n)))
5. free memory of old params.

I think this is main problem why we can't build complex details now.

Evil-Spirit, 2016-05-06:
Not so complex case, but ~1 fps while moving sketch point
[image]

jwesthues, 2016-05-07:
I'd profile to make sure the entity regeneration really is the bottleneck here; but agreed that you'll eventually need to make that selective, because that time will eventually dominate in a large enough model. Your Evil-Spirit@f6ac2e0 looks correct to me.

Evil-Spirit, 2016-05-07:
@jwesthues, 
I suspect about half of time is (unnecessary) mesh operations.

whitequark, 2016-05-07:
@Evil-Spirit, irc?

jwesthues, 2016-05-07:
@Evil-Spirit, I could easily believe that. Easy enough to measure, just add some debug prints with the time spent in each big step.

whitequark, 2016-05-07:
I'm rather looking forward to the time when we have a command line interface that allows us to measure speedup, both in the UI and sketch generation, across multiple dissimilar testcases.

Evil-Spirit, 2016-05-17:
Should I start doing this?

whitequark, 2016-05-17:
@Evil-Spirit what exactly?

Evil-Spirit, 2016-05-17:
The main usecase for optimization is: drawing in new workplane based on a lot of groups.
1. Figure out what is bottleneck (mesh operations on empty bodies or all-groups entitiy regeneration)
2. Avoid empty mesh ops or/and all the groups entity regeneration(do it just for requested groups).

whitequark, 2016-05-17:
After 3.0. Releasing 2.1 and getting hierarchical sketches in 3.0 is more important.

Evil-Spirit, 2016-05-17:
So, just remove 2.1. Probably you set it by mistake.

whitequark, 2016-10-10:
@Evil-Spirit I've committed Evil-Spirit@f6ac2e0 in master.

Evil-Spirit, 2016-10-24:
@whitequark,
@jwesthues 
This is not yet closed (the main theme which described in first message), please reopen.

I'll just leave it here:

before https://github.com/Evil-Spirit/solvespace/commit/60419e433d695b57d0862c3b8994ef2393c92fcd and https://github.com/Evil-Spirit/solvespace/commit/e1bd6e4ec2d8dcbc6501b2a27224de6fad6cab24
Generate::ALL (for bounding box) took 1136 ms
Generate::ALL took 3402 ms
Generate::DIRTY (for bounding box) took 1800 ms
Generate::DIRTY took 5731 ms

Generate::ALL (for bounding box) took 1130 ms
Generate::ALL took 3338 ms
Generate::DIRTY (for bounding box) took 1800 ms
Generate::DIRTY took 5450 ms

after https://github.com/Evil-Spirit/solvespace/commit/60419e433d695b57d0862c3b8994ef2393c92fcd and https://github.com/Evil-Spirit/solvespace/commit/e1bd6e4ec2d8dcbc6501b2a27224de6fad6cab24
Generate::ALL (for bounding box) took 1146 ms
Generate::ALL took 2651 ms
Generate::DIRTY (for bounding box) took 1783 ms
Generate::DIRTY took 3566 ms

Generate::ALL (for bounding box) took 1133 ms
Generate::ALL took 2632 ms
Generate::DIRTY (for bounding box) took 1797 ms
Generate::DIRTY took 3538 ms

Evil-Spirit, 2016-10-25:
More experiments.
https://github.com/Evil-Spirit/solvespace/commit/76d4ca0d77aa3201726e4743122f54f2b3408459
I don't know about correctness, but with it you can easily switch groups and dragging becomes pretty fast.

@jwesthues 
Need your opinion about correcness of this changes.

Evil-Spirit, 2016-11-10:
We can work with huge group-assemblies. Sketch is draggable.
100*6 = 600
[image]

whitequark, 2016-11-10:
Very interesting!

Evil-Spirit, 2016-11-10:
@whitequark,
I figured out what 2.1 is quite slow in generation, 2.0 was much faster.
I think we need turn back something like memmove because move_backward uses operator= and causes some reallocation work for stuff like std::string.

Evil-Spirit, 2016-11-10:
@whitequark,
IdList is dead slow because of Add() even with memmove. I think tree-based map will be better in this case.

whitequark, 2016-11-10:
> I think we need turn back something like memmove because move_backward uses operator= and causes some reallocation work for stuff like std::string.

Yup. You cannot use memmove with STL containers.

whitequark, 2016-11-17:
> I think we need turn back something like memmove because move_backward uses operator= and causes some reallocation work for stuff like std::string.

Actually, now that I think about it, I don't see why move_backward would do any reallocation. This is the purpose of moving: it is logically equivalent to a shallow swap operation. So the destination string grabs the buffer from the source string, and the source string grabs the (empty) buffer from the destination string, which should involve no allocation because it's 0 bytes.

Evil-Spirit, 2016-11-17:
@whitequark, I know, but tests shows the string coping take so much time. Anyway, I hope I've found some solution for IdList : https://github.com/Evil-Spirit/solvespace/commit/a38d7c0b5a772bd64b9ef53a86c6ba57527ca389

Evil-Spirit, 2016-11-17:
Before ______________________________________________
Generate::ALL (for bounding box) took 1001 ms
Generate::ALL took 15325 ms
Generate::DIRTY (for bounding box) took 1180 ms
Generate::DIRTY took 18719 ms
MaximumId______________________________________________
Generate::ALL (for bounding box) took 976 ms
Generate::ALL took 8327 ms
Generate::DIRTY (for bounding box) took 1150 ms
Generate::DIRTY took 12187 ms
Reserve_____________________________________________________________
Generate::ALL (for bounding box) took 1054 ms
Generate::ALL took 4582 ms
Generate::DIRTY (for bounding box) took 1017 ms
Generate::DIRTY took 3988 ms
Identity Checking________________________________________________________________
Generate::ALL (for bounding box) took 972 ms
Generate::ALL took 3160 ms
Generate::DIRTY (for bounding box) took 925 ms
Generate::DIRTY took 2881 ms
IdList indexing________________________________________________
Generate::ALL (for bounding box) took 416 ms
Generate::ALL took 2055 ms
Generate::DIRTY (for bounding box) took 374 ms
Generate::DIRTY took 1794 ms

Evil-Spirit, 2016-11-17:
@whitequark 
https://github.com/Evil-Spirit/solvespace/commits/opengl2

Evil-Spirit, 2016-11-17:
test file:
big-big-big-woodworking-asm.slvs

Evil-Spirit, 2016-11-18:
dragging big-big-big-woodworking-asm.slvs
IdList indexed_________________________________________________
Generate::DIRTY (for bounding box) took 374 ms
Generate::DIRTY took 722 ms
After DIRTY only regeneration________________________________
Generate::DIRTY (for bounding box) took 31 ms
Generate::DIRTY took 66 ms

whitequark, 2016-11-18:
@Evil-Spirit Wow! A 43 times improvement!

Evil-Spirit, 2016-11-18:
@whitequark, no, the last one is for dragging, the previous is for loading

Evil-Spirit, 2016-11-18:
actually the fixes for dragging improvement makes some regression for loading
Generate::DIRTY took 1794 ms -> ~3000ms.
But this is not so meaningful because this resulted in 10times faster dragging for the last group

Evil-Spirit, 2016-11-19:
@whitequark,
What do you thing about this https://github.com/Evil-Spirit/solvespace/commit/8027f3d74f91bff894f9264a5b41400ed27f6313
Do you know any container implementations where you can easily iterate and easily insert? I've implemented something what good for our case. If we need, we iterate by plain vector (cache-friendly). We also have additional vector for ordering items. When we making lookup for an item, we just do binary search in small cache-friendly array, then do derefernce once when found the result. If we need to insert item, we just append it for storage vector without any coping and reallocation (because this happens pretty rare, only when we have reach elemsAllocated). For inserting, we have only copy small array (we can memmove here)

whitequark, 2016-11-19:
This implementation looks pretty good to me.

Evil-Spirit, 2016-11-19:
@whitequark, I have also tried std::map and this is so bad... Insertion is fast, but iteration is hell slow.

whitequark, 2016-11-19:
That's about what I expect. std::map is usually an rbtree, which is bad for cache.

Evil-Spirit, 2016-12-21:
@whitequark,
rebased https://github.com/Evil-Spirit/solvespace-master/tree/performance

whitequark, 2017-01-12:
Merged "Optimize GroupInOrder even more"

Evil-Spirit, 2017-01-12:
@whitequark, reservers https://github.com/Evil-Spirit/solvespace-master/commit/c392fd2c894ea12f04c2d4236a12d7b2a78890df

Evil-Spirit, 2017-01-12:
@whitequark, reserves https://github.com/Evil-Spirit/solvespace-master/commit/2065500d12e3cd862c11fd4b954101935b87c98e

whitequark, 2017-01-14:
Merged "Reserves."

whitequark, 2017-02-10:
I don't think "IdList indexing" should be merged until all users of IdList use iterators.

whitequark, 2019-05-23:
Moved to https://github.com/solvespace/solvespace/issues/430.

closed 2019-05-23 by whitequark