whitequark/solvespace #148: Three.js exported files should not rely on network

whitequark, 2016-05-06:


whitequark, 2016-05-18:
Fixed in 432e768. @cr1901, see what the plan was all along?

closed 2016-05-18 by whitequark

cr1901, 2016-05-18:
@whitequark I do see it now! No need to update either, so the viewer (sans not-yet-found bugs on my end) will always "just work", as long as WebGL is supported in browsers :); I assume you'll update if there's security vulns.

Also, TIL you can embed arbitrary resources (makes sense, I just never needed to, so I didn't know).

whitequark, 2016-05-18:
> I assume you'll update if there's security vulns.

I don't see how this code would have any, ever. It does not read or write data.

> TIL you can embed arbitrary resources

I just implemented that a few days ago.