whitequark/solvespace #156: changing import paths after copying

tgirod, 2016-05-12:
Let's say you have this:

```
v1/piece1.slvs
v1/piece2.slvs
v1/assembly.slvs # imports piece1 and piece2
```
now, you copy the `v1` folder and its content to a `v2` folder.

```
v2/piece1.slvs
v2/piece2.slvs
v2/assembly.slvs # imports piece1 and piece2
```
If you open `v2/assembly.slvs`, you will notice that the v1 pieces are imported, rather than the v2 pieces. I guess the import path auto-correction did some magic here. I think it would be more sensible to import v2 pieces, right ? Moreover, it is not possible to change those paths from solvespace, you have to edit the file manually.

whitequark, 2016-05-12:
That's actually an inverse case. Relative paths would only be used when the file can not be found at the recorded absolute path. I'm not sure if there's one behavior that's always preferable here... @jwesthues ?

jwesthues, 2016-05-13:
I'm pretty sure SolveSpace looks at absolute first (and then relative) only because I coded it with absolute paths only first, and then realized later that relative might be useful sometimes too. No objection to swapping the precedence when both exist.

Evil-Spirit, 2016-05-13:
It looks logical to swap the paths: firstly look at relative, then in absolute.

whitequark, 2016-05-17:
Fixed in 0d7aa0a.

closed 2016-05-17 by whitequark