whitequark/solvespace #157: Printing

whitequark, 2016-05-12:
Let's have a File → Print... option ourselves, with preview and everything. Simple matter of OS integration.

Evil-Spirit, 2016-08-13:
http://www.codeproject.com/Articles/8916/Printing-Architecture

whitequark, 2016-08-13:
And what about print dialogs?

Evil-Spirit, 2016-08-13:
@whitequark, We can call for standart printing dialog to choose printer, but seems what preview we have do our own. Also, I found info how to print using PostScript itsef, not it's pdf version. We already have PostScript output, so this is not a big problem.
http://stackoverflow.com/questions/2240243/how-do-i-do-print-preview-in-win32-c

whitequark, 2016-08-13:
That answer says nothing about the dialogs.

Evil-Spirit, 2016-08-15:
https://msdn.microsoft.com/en-us/library/windows/desktop/ms646829.aspx#print_property

whitequark, 2016-10-14:
Moved to solvespace/solvespace

closed 2016-10-14 by whitequark