whitequark/solvespace #159: GTK+3 build fails

prokoudine, 2016-05-14:
Building on Fedora 23 with GTK+ 3.18.9 fails:

[ 51%] Building CXX object src/CMakeFiles/solvespace.dir/gtk/gtkmain.cpp.o
/home/alex/soft/cad/solvespace/src/gtk/gtkmain.cpp: In member function 'void SolveSpace::EditorOverlay::start_editing(int, int, int, bool, int, const string&)':
/home/alex/soft/cad/solvespace/src/gtk/gtkmain.cpp:394:16: error: 'class Gtk::Entry' has no member named 'override_font'
         _entry.override_font(override_font_desc);
                ^
/home/alex/soft/cad/solvespace/src/gtk/gtkmain.cpp:409:37: error: 'class Gtk::Entry' has no member named 'get_style_context'
         Gtk::Border border = _entry.get_style_context()->get_padding();
                                     ^
/home/alex/soft/cad/solvespace/src/gtk/gtkmain.cpp:410:33: error: 'Gtk::Border {aka struct _GtkBorder}' has no member named 'get_left'
         move(_entry, x - border.get_left(), y - border.get_top());
                                 ^
/home/alex/soft/cad/solvespace/src/gtk/gtkmain.cpp:410:56: error: 'Gtk::Border {aka struct _GtkBorder}' has no member named 'get_top'
         move(_entry, x - border.get_left(), y - border.get_top());
                                                        ^
/home/alex/soft/cad/solvespace/src/gtk/gtkmain.cpp: In function 'int SolveSpace::DeltaYOfScrollEvent(GdkEventScroll*)':
/home/alex/soft/cad/solvespace/src/gtk/gtkmain.cpp:483:26: error: 'GdkEventScroll {aka struct _GdkEventScroll}' has no member named 'delta_y'
     int delta_y = event->delta_y;
                          ^
/home/alex/soft/cad/solvespace/src/gtk/gtkmain.cpp: In function 'std::__cxx11::string SolveSpace::ConvertFilters(std::__cxx11::string, const SolveSpace::FileFilter*, Gtk::FileChooser*)':
/home/alex/soft/cad/solvespace/src/gtk/gtkmain.cpp:1061:48: error: 'create' is not a member of 'Gtk::FileFilter'
         Glib::RefPtr<Gtk::FileFilter> filter = Gtk::FileFilter::create();
                                                ^
/home/alex/soft/cad/solvespace/src/gtk/gtkmain.cpp:1084:35: error: no matching function for call to 'Gtk::FileChooser::add_filter(Glib::RefPtr<Gtk::FileFilter>&)'
         chooser->add_filter(filter);
                                   ^
In file included from /usr/include/gtkmm-2.4/gtkmm/filechooserdialog.h:31:0,
                 from /home/alex/soft/cad/solvespace/src/gtk/gtkmain.cpp:36:
/usr/include/gtkmm-2.4/gtkmm/filechooser.h:910:8: note: candidate: void Gtk::FileChooser::add_filter(const Gtk::FileFilter&)
   void add_filter(const FileFilter& filter);
        ^
/usr/include/gtkmm-2.4/gtkmm/filechooser.h:910:8: note:   no known conversion for argument 1 from 'Glib::RefPtr<Gtk::FileFilter>' to 'const Gtk::FileFilter&'
/home/alex/soft/cad/solvespace/src/gtk/gtkmain.cpp:1086:39: error: no matching function for call to 'Gtk::FileChooser::set_filter(Glib::RefPtr<Gtk::FileFilter>&)'
             chooser->set_filter(filter);
                                       ^
In file included from /usr/include/gtkmm-2.4/gtkmm/filechooserdialog.h:31:0,
                 from /home/alex/soft/cad/solvespace/src/gtk/gtkmain.cpp:36:
/usr/include/gtkmm-2.4/gtkmm/filechooser.h:951:8: note: candidate: void Gtk::FileChooser::set_filter(const Gtk::FileFilter&)
   void set_filter(const FileFilter& filter);
        ^
/usr/include/gtkmm-2.4/gtkmm/filechooser.h:951:8: note:   no known conversion for argument 1 from 'Glib::RefPtr<Gtk::FileFilter>' to 'const Gtk::FileFilter&'
/home/alex/soft/cad/solvespace/src/gtk/gtkmain.cpp: In member function 'void SolveSpace::TextWidget::set_cursor_hand(bool)':
/home/alex/soft/cad/solvespace/src/gtk/gtkmain.cpp:1271:32: error: 'create' is not a member of 'Gdk::Cursor'
             gdkwin->set_cursor(Gdk::Cursor::create(type));
                                ^
/home/alex/soft/cad/solvespace/src/gtk/gtkmain.cpp: In constructor 'SolveSpace::TextWindowGtk::TextWindowGtk()':
/home/alex/soft/cad/solvespace/src/gtk/gtkmain.cpp:1323:47: error: no matching function for call to 'SolveSpace::TextWidget::TextWidget(Gtk::Adjustment*)'
                       _overlay(_widget), _box() {
                                               ^
/home/alex/soft/cad/solvespace/src/gtk/gtkmain.cpp:1258:5: note: candidate: SolveSpace::TextWidget::TextWidget(Glib::RefPtr<Gtk::Adjustment>)
     TextWidget(Glib::RefPtr<Gtk::Adjustment> adjustment) : _adjustment(adjustment) {
     ^
/home/alex/soft/cad/solvespace/src/gtk/gtkmain.cpp:1258:5: note:   no known conversion for argument 1 from 'Gtk::Adjustment*' to 'Glib::RefPtr<Gtk::Adjustment>'
src/CMakeFiles/solvespace.dir/build.make:295: recipe for target 'src/CMakeFiles/solvespace.dir/gtk/gtkmain.cpp.o' failed
make[2]: *** [src/CMakeFiles/solvespace.dir/gtk/gtkmain.cpp.o] Error 1
CMakeFiles/Makefile2:316: recipe for target 'src/CMakeFiles/solvespace.dir/all' failed
make[1]: *** [src/CMakeFiles/solvespace.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

whitequark, 2016-05-14:
> /usr/include/gtkmm-2.4

You aren't building with gtk3.

Have you passed `-DGUI=gtk3` to cmake?

whitequark, 2016-05-14:
Actually, scratch that. The code in `gtkmain.cpp` that is built is for GTK3, yet the gtkmm version corresponds to gtk2. I'm not sure how you even got into this state. Try removing your build tree and starting over.

prokoudine, 2016-05-14:
I ran ccmake and changed gtk2 to gtk3, then configured and ran make. Anyway, turns out, I didn't have gtkmm3 installed. Sorry to have bothered you for nothing :)

closed 2016-05-14 by prokoudine