whitequark/solvespace #162: Export interactive PDF (ideas)

Symbian9, 2016-05-17:
Interactive PDF forms and other futures can be very usefull for creating reports and technical drawings.
* https://leadertechinc.com/news/slot-lok-technical-drawings-released/
* http://www.lymnassociates.com/technical-drawing.html
[image]

Now few 3rd party tools can create some of this futures:
* http://www.techrepublic.com/blog/tr-dojo/create-interactive-pdf-documents-with-scribus/

But I think that it would be usefull add some option for export interactive PDF in SolveSpace.

<b>In this issue I propose discuss possibility of export PDF with interactive futures</b>:
* show or hide comments or other elements when cursor hover special region on PDF page - http://indesignsecrets.com/showing-and-hiding-objects-in-interactive-pdf.php
* export 3D model directly in PDF (U3D format) - https://en.wikipedia.org/wiki/Universal_3D

More ideas coming soon! (this message would be updated)

Evil-Spirit, 2016-05-17:
Can you explain why?

Evil-Spirit, 2016-05-17:
> U3D format

One more "simple", "universal" format (specifiaction consist of 237 pages).

Symbian9, 2016-05-17:
@Evil-Spirit, there are many reasons and here are some of them:
1) PDF is one of the most popular exchange format for technical drawing reports (after DXF, DWG and few other)
2) PDF readers are one of the application that already preinstalled or firstly instaled on most of users side PC.
3) Some companies not use any CAD software (and not allow install 3rd party CAD), but allow use of PDF documentation.
4) Interactive PDF already used in many university as form of testing papers.
5) Three.js with WebGL can't work on PC with low RAM and GPU, where interactive PDF can work normally.

Also PDF itself is multipage document, that for reports can be greate. Most of well known complex CAD can produce multipage reports and export to PDF. I hope SolveSpace in future would be complex CAD too.

Evil-Spirit, 2016-05-17:
What's wrong with the current PDF export for technical drawings?

Evil-Spirit, 2016-05-17:
1)2)3) We already have the PDF export;
4) We don't make the program for tesing of the students.
5) ???? WHAT ????

Evil-Spirit, 2016-05-17:
So, why you have mentioned "ineractive", "u3d".
You just have to create "make pdf more userful for technical drawings".
There is no need in silence links, just write the features you REALLY need, like

* multipage pdf
* whatever

You just overload us the unrelated information and the some links (i even don't know what I must figure out from looking at)

Symbian9, 2016-05-17:
@Evil-Spirit, U3D Format is only one of many futures of PDF, but I not say that it "must" be presented in SolveSpace. Think, there can be used some pseudo 3D futures in PDF using JS and CSS (now I'm in search of 3rd party libraries and tricks for that).

For me more interesting is interactive 2D techical drawing reports with interactive comments, switched layers and multipage export with some additional information. For example, resulted PDF can have drawing on 1st page and textual specification in table form on 2nd page or in other order. Think it can be released by adding comments futures for groups inside text browser and few other option for creating specification pages in PDF automaticaly (especially for assembled models). 
[image]

whitequark, 2016-05-18:
Out of scope for me. It's open-source; if you want it, implement it.

whitequark, 2016-05-18:
Also, U3D effectively only works with proprietary Acrobat, so that's completely useless.

closed 2016-05-18 by whitequark

Symbian9, 2016-05-18:
@whitequark, I propose 3D implementing using PRC (as alternative to U3D) with next opensource thing
* http://asymptote.sourceforge.net/

whitequark, 2016-05-18:
That's still limited to Adobe Reader: https://en.wikipedia.org/wiki/PRC_(file_format)#Viewer