whitequark/solvespace #165: Allow use internal "plotter-style" font for TrueType Text.

Symbian9, 2016-05-18:
Please, make it possible use internal SolveSpace font for TrueType text blocks.

P.S.: I start think about it when cant rotate comment text (maybe you soon can allow rotating future for comments too).

Evil-Spirit, 2016-05-18:
rotating for comments is supported, you need to create custom style and assign it to comment. you can choose text options for custom style.

Symbian9, 2016-05-18:
Is "TrueType font text" allow plotter fonts? (single line without fill)

whitequark, 2016-05-18:
@Symbian9 The TrueType renderer does not care about filling, it is implicit in TrueType, with the same behavior as SolveSpace sketches (outermost space is unfilled and then it alternates). So if you use a plotter-style font you'll get a broken sketch, because you can't do what you want with entities at all.

whitequark, 2016-05-18:
What is the purpose of using plotter font for text entities?

Evil-Spirit, 2016-05-18:
@Symbian9, there is no point to use true type as plotter. true type is for building 3d font objects, so it need to have closed contours for each glyph. for the poltter there is comment.

closed 2016-05-18 by whitequark