whitequark/solvespace #166: Align text

Symbian9, 2016-05-18:
Is it possible add align option for "TrueType font" and "Comment" text.

Allign types:
* left
* center
* right

Also, what about "vertical text"?

Evil-Spirit, 2016-05-18:
already implemented

Symbian9, 2016-05-18:
where implemented? In latest release (2.1~e129755d) cant find this functionality

Evil-Spirit, 2016-05-18:
please, provide information about real use cases.

whitequark, 2016-05-18:
For text comments, both horizontal and vertical alignment is already implemented. It makes sense because the comment is aligned using a single origin point, which is not enough to completely define it.

For TrueType text, vertical alignment doesn't make any sense at all because it is specified using two points, baseline and ascender position. You can do trivial construction to get a point centered between two.

Horizontal alignment does make sense.

Evil-Spirit, 2016-05-18:
@whitequark, makes sense additional points for horizontal length of true type text (to have ability constraint it)

whitequark, 2016-05-18:
That might be even better. But I'm not sure how that interacts with the solver.

Evil-Spirit, 2016-05-18:
There is no problem with solver because each entity can generate some equations (now used for acrs). We can add new point and constrain horizontal vector to be perpendicular with vertical.

Evil-Spirit, 2016-05-18:
@jwesthues, what do you think about it?

whitequark, 2016-05-18:
If we designate bottom-left point as origin, then the len(origin-y):len(origin-x) ratio is constrained by the text being rendered. I'm not sure how to represent it.

whitequark, 2016-05-18:
I guess we could ask FreeType to render the text at an arbitrary size "1", which will give us the ratio of y to x size. With hinting turned off (as it is now) that shouldn't cause any issues after scaling up...

jwesthues, 2016-05-18:
Like generate two more points, constrained with the entity equations to lie at the vertices of a rectangle with the aspect ratio that you get by rendering the text at arbitrary size? That should work nicely with the solver, and remain backwards-compatible as long as you don't renumber the original two.

whitequark, 2016-05-26:
Hm, one problem I see is what happens when the font isn't available--we probably should record the aspect ratio, and not do anything to it during regeneration when the font isn't found, or sketches will break.

whitequark, 2016-10-11:
I've implemented most of this in the `horz-text-pt` branch, but it doesn't pass 2.0 migration tests, and also the horizontal point can get inverted.

whitequark, 2016-10-14:
@jwesthues So looks like this is a backwards compatibility hazard, as in an underconstrained sketch, the newly added point starts at origin (I think) in sketches from 2.x, and the solver would converge to something else entirely. E.g. from our tests:

expected:
[image]
actual:
[image]

(disregard the sign bug, I haven't yet applied @Evil-Spirit's recommendation)

How bad do you think this is? I feel it's not bad but I may be missing something.

Evil-Spirit, 2016-10-14:
@whitequark, solver find some near solution. to force it to not change anything at first solve we must suggest already converged parameters. So, for just created text we need to specify all the points coordinates to right positions. Somehow we can find out what this point don't exist before and force it coordinates to right positions

whitequark, 2016-10-14:
So the aspect ratio is calculated, I think, before the entities corresponding to those points would be created. So if we detect that there are no such entities, in the code in Request::Generate, we can probably place them if we know the aspect ratio and u/v vectors.

jwesthues, 2016-10-14:
Agreed that you should be able to fill a correct initial guess when the loaded file doesn't mention the new point.

Evil-Spirit, 2016-10-18:
@whitequark, @jwesthues,
rewritten equations in https://github.com/whitequark/solvespace/commit/479093bfa422c079b546e280d4ca544f231ee2c9

Evil-Spirit, 2016-10-19:
@whitequark, @jwesthues 
Implemented some way to check whether parameter already guessed or just generated.

Evil-Spirit, 2016-10-19:
here https://github.com/whitequark/solvespace/commit/f371472b4e172097f76d4c27ffc440b212d17747

Evil-Spirit, 2016-10-19:
@jwesthues, need your opinion about moving 
`// Set the initial guesses for all the params`
into Solve() body

jwesthues, 2016-10-20:
@Evil-Spirit I don't understand why this change involves the solver. If you know the text's aspect ratio and the old two points, then why can't you compute a correct initial position for the new third point before invoking the solver?

Evil-Spirit, 2016-10-20:
@jwesthues 
You mean we can create function EntityBase::ApplyMissingGuess before all the solver work?
I thought about it, but fact what we have all the information inside GenerateEquations was my reason not to do it such way. But this causes problem what we already copied all the guesses and changing SK.params meaning noting, so I have to move copy of initial guesses

jwesthues, 2016-10-20:
@Evil-Spirit Can't you just do it once, immediately after loading the file? There's no need to ever touch the new point after that, and that keeps the ugly backwards-compatibility hack contained.

Evil-Spirit, 2016-10-20:
@jwesthues,
This is just prototype, I was thought not only about working this as legacy loading hack, but also for brand-new requests (which also have bad initial guess causes unpredictable behaviour)

jwesthues, 2016-10-20:
@Evil-Spirit If you fill a reasonable value for all three points when you first create the request, then with the first fixed and the second dragged, shouldn't the solver smoothly compute the third?

Evil-Spirit, 2016-10-20:
@jwesthues, yes, but I mean the first creation here. After solver will compute it.

whitequark, 2016-10-20:
> There's no need to ever touch the new point after that, and that keeps the ugly backwards-compatibility hack contained.

@Evil-Spirit I agree with Jonathan here, this shouldn't go into the solver internals... you should be able to do everything from the existing numeric coordinates just after the file is loaded, no?

Evil-Spirit, 2016-10-20:
@whitequark, We need to understad which parameters was loaded from file and which of them was created. We can't do it without regeneration, so we can do it just befroe solve in separated function EntityBase::ApplyMissingGuess

whitequark, 2016-10-20:
@Evil-Spirit Wouldn't the point be missing just after loading the sketch, but before generation? So you can check that with `FindByIdNoOops`

Evil-Spirit, 2016-10-20:
@whitequark,
Entities doesn't load from normal files, just for imported. It always regenerated just after loading. So we have to analyze which parameters was loaded and which was just created by generation.
@jwesthues 
May be there is other better ways?

Evil-Spirit, 2016-10-20:
@whitequark,
@jwesthues 
The second iteration https://github.com/Evil-Spirit/solvespace/commit/44e690398bbeba23e1cf5bf932103d3c6662346b

whitequark, 2016-11-02:
Merged.

closed 2016-11-02 by whitequark