whitequark/solvespace #167: Hide construction (points & lines)

Symbian9, 2016-05-18:
Propose allow hide all "green" elements. For now its possible hide only points (that itself is "constructed" element), but lines and curves (that switched to "CONSTRUCTION") not yet possible hide.
So, think it would be usefull use already presented "Hide points" tab in text window for hide all dots and construction elements in one click and rename this tab to "Hide construction"

Evil-Spirit, 2016-05-18:
You can assign custom style with ability to change "show these objects on screen". Checking/unchecking this option can emulate this behaviour. I think it will be useful to unlock all the style options even for default styles, so in future you can do it without any custom styles.

Symbian9, 2016-05-18:
@Evil-Spirit, "CONSTRUCTION" is a type, not a style. I agree with what you say about styles, but anyway propose replace "Hide points" with "Hide construction" button (for hiding all "constructed" elements: points, lines, curves, arcs etc.)

whitequark, 2016-05-18:
Points are not construction elements though.

whitequark, 2016-05-18:
> I think it will be useful to unlock all the style options even for default styles, so in future you can do it without any custom styles.

Most of them are never used, which is rather confusing. (Even in the *restricted* list of options, most of them are never used.) That'll have to wait until the new styling system.

Evil-Spirit, 2016-05-18:
@whitequark, ok, we have to change styling systeem completely.

whitequark, 2016-08-13:
@jwesthues Opinion on changing the behavior of the "Show/hide points" button to "Show/hide points and construction entities" ?

jwesthues, 2016-08-13:
Why should construction entities track with points vs. normals or workplanes?

whitequark, 2016-08-13:
Hm, good point. But I don't like the idea of adding yet another button for construction entities. The text window toolbar is too crowded as it is.

whitequark, 2016-08-13:
@Symbian9 I agree with @jwesthues, there is no reason to change the behavior of "Show/hide points" button like this.

closed 2016-08-13 by whitequark