whitequark/solvespace #168: "Text Window" or "Browser"?

Symbian9, 2016-05-18:
I sew this confuse long time: when you go to "View" menu you will see "Show Text Window", but this window has header "Browser".
I propose rename both this to one - "Property Browser"

whitequark, 2016-05-18:
@jwesthues? Not sure what the original intent was.

jwesthues, 2016-05-18:
I started with "text window", but changed (incompletely) to "browser". The rename seems reasonable to me.

whitequark, 2016-05-18:
Done in b55e096fefd2.

closed 2016-05-18 by whitequark

Symbian9, 2016-05-18:
@whitequark, think, in code there is also an issue that need to rename some , for example <code>MNU_SHOW_TEXT_WND</code>.
I think all strings that include <code>..._TEXT_WND...</code> must be replaced with something that related to header of this window.