whitequark/solvespace #170: Icons are not rendered on Windows

whitequark, 2016-05-18:
This is some weird Windows-specific OpenGL problem. They're all loaded correctly (verified in a debugger) but for some reason render as black rectangles.

Evil-Spirit, 2016-05-18:
Ok, will take a look into this.

Evil-Spirit, 2016-05-20:
Where can i see it?

Evil-Spirit, 2016-05-20:
@whitequark, can you provide more information?

whitequark, 2016-05-20:
This is what I get when building latest master and running on Windows 7 in VirtualBox:
[image]

Evil-Spirit, 2016-05-20:
@whitequark, I have no such problem (Windows 7 x64)

Evil-Spirit, 2016-05-20:
Is this problem only for virtual box?

whitequark, 2016-05-20:
I don't know, I don't have native Windows.

whitequark, 2016-05-20:
If it's a problem just for VirtualBox then it probably still means misuse of GL that needs to be fixed.

Evil-Spirit, 2016-05-20:
> glBindTexture(GL_TEXTURE_2D, TEXTURE_DRAW_PIXELS);
You have to call glGenTextures to generate id instead of using TEXTURE_DRAW_PIXELS

whitequark, 2016-05-20:
But that works elsewhere and also worked before, no?

Evil-Spirit, 2016-05-20:
I think this was before some specific moment.
the gl spec says
`You must use glGenTextures to generate a set of new texture names.`

Evil-Spirit, 2016-05-20:
OpenGL always shows good behaviour on the corrupted/wrong input or wrong using of the functions.
When I write my own game projects, such misusing works for a years. Until the some specific moment when memory segments becomes the different or until driver updating.

whitequark, 2016-05-20:
Yeah you are right. Since this works without Virtualbox let's just fix this when rewriting the rendering code, which should be soon anyway.

Evil-Spirit, 2016-05-21:
I think we can leave it.

whitequark, 2016-07-31:
@Evil-Spirit This is still a problem, and using glGenTextures does not fix it. I'll need you to install Windows in VirtualBox and figure out why it fails.

Evil-Spirit, 2016-08-16:
Have not get success to reproduce this. 
I've tried 
* Under VirtualBox for:
  * WinXP SP3
  * Windows 7 x86 runned on two different laptops (GF650m+Corei7, Intel Atom Z8300)
* Also tried VMWare with no success.
* Without virtualization under:
  * Windows 7 x64
  * Windows 10 x64.

whitequark, 2016-08-17:
@Evil-Spirit I can still reproduce it. I get `glGetError() == 0x502` messages. Is there some way to get a more verbose explanation on Windows?

whitequark, 2016-08-17:
@Evil-Spirit I got a GL API trace with apitrace 
[trace.zip](https://github.com/whitequark/solvespace/files/422482/trace.zip). Probably won't help you since apitrace doesn't record errors, only GL calls. But anyway, this is what it shows:
[image]

How do I get a more useful error out of Windows?

whitequark, 2016-08-17:
@Evil-Spirit Also, did you enable any acceleration in vbox? I am running without either 3d or 2d acceleration enabled.

Evil-Spirit, 2016-08-17:
@whitequark, I have tested both with and without acceleration. There all good parameters passed, but could you try to pass 16x16 instead of 24x24 for width and height? May be OpenGL excepted POT texture size?

whitequark, 2016-08-17:
@Evil-Spirit Bingo! What sort of fucking braindead restriction is that.

whitequark, 2016-08-17:
This is apparently how GL1.1 works. https://www.opengl.org/registry/specs/ARB/texture_non_power_of_two.txt

whitequark, 2016-10-09:
Fixed in master.

closed 2016-10-09 by whitequark