whitequark/solvespace #172: Major memory leak on OS X makes application unusable

follower, 2016-05-18:
When I tried out the latest nightly (https://github.com/whitequark/solvespace/releases/tag/nightly.2016-05-11) of solvespace it quickly became unusable even for simple drawings.

I eventually noticed that over a short amount of time it would allocate 6-8 *gigabytes* of memory! This greatly impacts system performance.

After further investigation I noticed that the memory allocation would increase each time the mouse pointer moved over any item, with the exception of the background of the main window and context menus.

I used the Xcode "Instruments" tool which revealed that multiple large allocations were being made each time I moved the pointer over an item. The call stack pointed to [`drawRect`](https://github.com/whitequark/solvespace/blob/45514849e20a94d3c103790446a84d4f677f3167/src/platform/cocoamain.mm#L202) as the culprit.

It turns out the size of the memory allocated is related to the size of the main window.

The associated call stack is:

```
   0 libsystem_c.dylib malloc
   1 CoreGraphics CGSImageDataHandleCreate
   2 CoreGraphics img_data_lock
   3 CoreGraphics CGSImageDataLock
   4 libRIP.A.dylib ripc_AcquireImage
   5 libRIP.A.dylib ripc_DrawImage
   6 CoreGraphics CGContextDrawImage
   7 solvespace -[GLViewWithEditor drawRect:]
   8 AppKit -[NSView(NSInternal) _recursive:displayRectIgnoringOpacity:inContext:topView:]
   9 AppKit -[NSView(NSLayerKitGlue) _drawViewBackingLayer:inContext:drawingHandler:]
  10 AppKit -[NSView(NSLayerKitGlue) drawLayer:inContext:]
  11 QuartzCore -[CALayer drawInContext:]
  12 QuartzCore CABackingStoreUpdate_
  13 QuartzCore CA::Layer::display_()
  14 QuartzCore CA::Layer::display_if_needed(CA::Transaction*)
  15 QuartzCore CA::Layer::layout_and_display_if_needed(CA::Transaction*)
  16 QuartzCore CA::Context::commit_transaction(CA::Transaction*)
  17 QuartzCore CA::Transaction::commit()
  18 AppKit -[NSView(NSLayerKitGlue) _drawRectAsLayerTree:]
  19 AppKit -[NSView _drawRect:clip:]
  20 AppKit -[NSView _recursiveDisplayAllDirtyWithLockFocus:visRect:]
  21 AppKit -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:]
  22 AppKit -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:]
  23 AppKit -[NSThemeFrame _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:]
  24 AppKit -[NSView _displayRectIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:]
  25 AppKit -[NSView displayIfNeeded]
  26 AppKit _handleWindowNeedsDisplayOrLayoutOrUpdateConstraints
  27 AppKit __83-[NSWindow _postWindowNeedsDisplayOrLayoutOrUpdateConstraintsUnlessPostingDisabled]_block_invoke_01208
  28 CoreFoundation __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__
  29 CoreFoundation __CFRunLoopDoObservers
  30 CoreFoundation __CFRunLoopRun
  31 CoreFoundation CFRunLoopRunSpecific
  32 solvespace SolveSpace::PaintGraphics()
  33 solvespace SolveSpace::GraphicsWindow::HitTestMakeSelection(SolveSpace::Point2d)
  34 solvespace SolveSpace::GraphicsWindow::MouseMoved(double, double, bool, bool, bool, bool, bool)
  35 solvespace -[GraphicsWindowView mouseMoved:]
  36 AppKit -[NSWindow sendEvent:]
  37 AppKit -[NSApplication sendEvent:]
  38 AppKit -[NSApplication run]
  39 solvespace main
  40 solvespace start
```

I assume that the image needs to be deallocated before being redrawn or something similar?

Potentially helpful references:
 * https://developer.apple.com/library/mac/documentation/GraphicsImaging/Conceptual/drawingwithquartz2d/dq_context/dq_context.html#//apple_ref/doc/uid/TP30001066-CH203-BCICJBGG ("Listing 2-6  Drawing to a bitmap graphics context" final steps seem to be what's needed?)
 * http://stackoverflow.com/questions/6729351/cgcontextdrawimage-eating-my-memory-any-memory-leak
 * https://developer.apple.com/library/mac/documentation/GraphicsImaging/Reference/CGContext/#//apple_ref/c/func/CGContextDrawImage
 * http://forum.sparrow-framework.org/topic/memoryleak-in-screenshot-funktion

whitequark, 2016-05-18:
Let me guess, do you have your display configured as Retina/2x?

follower, 2016-05-18:
On 19 May 2016 at 01:47, whitequark <notifications@github.com> wrote:

> Let me guess, do you have your display configured as Retina/2x?
>
Not Retina/2x, no. (Sorry, realised I hadn't specified OS/host details:
It's a pre-Retina Mac Book Pro running OS X 10.8.5.)

whitequark, 2016-05-18:
Ah, alright. Now that I look at the code it should manifest either way.

whitequark, 2016-05-18:
Ok, how do I... run Instruments? I can't figure that out. (Don't laugh, I never actually *use* OS X, I literally installed it with the sole purpose of making this port.)

whitequark, 2016-05-18:
Nevermind, figured it out. Also, yeah, this leak is quite dramatic.

follower, 2016-05-18:
On 19 May 2016 at 02:00, whitequark <notifications@github.com> wrote:
> Ok, how do I... run Instruments? I can't figure that out.
Not sure on a more recent OS/Xcode versions, but I searched for
"Instruments" via Spotlight and it found it within the Xcode
application folder. I assume you can run it from within Xcode somehow?
I don't actually use Xcode for development, just had a vague
recollection of the Instruments tool (which I think is based on
dtrace?)

> (Don't laugh, I never actually use OS X, I literally installed it with the
> sole purpose of making this port.)
For the record, I greatly appreciate you doing so. :)

whitequark, 2016-05-18:
Fixed in 63398ef3; Instruments show no more leaks in SolveSpace. This will be backported to 2.1 but in the meantime I guess you could just build `master`...

closed 2016-05-18 by whitequark