whitequark/solvespace #173: locking the camera on the z-axis ?

tgirod, 2016-05-18:
When drawing a shape, I often find myself jiggling it in 3D in order to turn it around.

I think it would be nice to have an option to lock the camera movement on the z-axis, so you can move around the shape while still preserving the vertical orientation (I think sketchup works like this).

whitequark, 2016-05-18:
I think that's in `wishlist.txt`... https://github.com/whitequark/solvespace/blob/master/wishlist.txt#L6

tgirod, 2016-05-18:
@whitequark ok, good!

tgirod, 2016-05-21:
#182 is a proposal for this.

whitequark, 2016-05-22:
If I understand correctly what you're doing, doesn't the following work:
  * making the camera parallel to XY plane, or any other one you would like;
  * rotating around the Z axis by holding Ctrl or Cmd while rotating?

tgirod, 2016-05-23:
There is just a bit more:

* making the camera parallel to the XY plane
* rotating around the Z axis to move horizontally
* rotating around projRight to move vertically

For a more general case, we could replace the XY plane with an arbitrary plane P and the Z axis with the normal vector of P.

if you think this approach is better, I can work on this.

whitequark, 2016-05-23:
No, I mean, that's already implemented.

whitequark, 2016-05-23:
So you can make the camera parallel to the plane you want using F2 or W, and then rotate while holding Ctrl.  Not rotating around projRight, it seems...

tgirod, 2016-05-23:
The current behaviour associated to the Ctrl key is completely different from what I'm aiming at. Maybe we were not talking about the same thing from the start.

What you describe is make the camera parallel to a plane (projRight *and* projUp are parallel to the plane), then rotate around the normal vector of this plane. It's like locking the camera on the north/south pole and watch the planet turn on its axis.

What I propose is to make projRight only parallel to a plane (XY plane for now). To continue my comparison, that would allow the camera to move all around the planet, while staying parallel to the equator (projRight parallel to the XY plane) and pointing to the north pole (projUp.z positive).

whitequark, 2016-05-24:
Oh, I see what you mean now. To be honest I'm not sure how broadly useful it is or how to best implement controls for it... hmm.

tgirod, 2016-05-24:
From my (limited) experience with CAD softwares, I find this camera movement much more intuitive, albeit more constrained. Well, that's just, like, my opinion, man. But my guts say that's probably the reason why it is the default behaviour in sketchup.

Anyway, if I were to do it, I would probably go like this:

1. don't replace the current behaviour - activate it through a configuration option
2. lock on the XY plane only, rather than an arbitrary plane, to make the movement as dumb and predictable as possible (that's the whole point)
3. like the regular mode, ctrl+middle click allows you to roll the camera - that way, it is possible to orient your volume the way you want before creating a new woking plane.
4. if the camera is not parallel to the XY plane (because of point 3), correct the orientation smoothly, like F2 and F3 before moving.

whitequark, 2016-05-24:
Okay, I will check out sketchup, to begin with.

whitequark, 2016-11-27:
@tgirod I've looked at sketchup and your idea's fine. I've migrated the issue to https://github.com/solvespace/solvespace/issues/123.

closed 2016-11-27 by whitequark