whitequark/solvespace #177: Constrain curve connected with other to be perpendicular

Symbian9, 2016-05-20:
Currently its only possible constrain curve connected with line as tangent (parallel). 
But , is it possible constrain them as perpendicular?

Evil-Spirit, 2016-05-20:
@Symbian9 , You can draw some construction and get what you want.

Symbian9, 2016-05-20:
@Evil-Spirit, for complex sketch adding too much additional constructing elements is bad technology. Also, it will increase file size because one additional variable is smaller then one more additional construction element in SLVS code specification.
Also, using construction line for so simple operation need two three operation instead if this issue would be.
For example doing this operation now need:
* create line
* create curve
* create additional line
* constrain perpendicular additional line with main line
* constrain parallel curve with additional line, that constrained perpendicular to main line.

So, this future can save many hours of work.
Also, parallel and perpendicular are most used constraints on drawings, thats why, think, them must work without needing make any construction elements for doing those tasks.

Evil-Spirit, 2016-05-20:
@Symbian9, I think the tanget for curves/segments is quite common workflow, but curve perpendicularity is no so common. Can you demonstrate a real example where you want to have a big number of such constraints?

Symbian9, 2016-05-20:
Evil-Spirit, one of common example is lamps design
[image]

Also, it usefull for many tubes connection on engines (on cars, aircraft etc.)
[image]

And many other examples in hydraulic systems, electricity 3D plans (inside house), architecture elements...

Evil-Spirit, 2016-05-20:
@Symbian9, I don't see any perpendicularity here. How can I calculate the number?

whitequark, 2016-05-20:
@Symbian9 Neither of your examples is anywhere close to being possible to do in SolveSpace since they all require, at least, lofting.

closed 2016-05-20 by whitequark

Symbian9, 2016-05-20:
@whitequark, is "Lofting" on your task list?

whitequark, 2016-05-20:
Not currently. We do not have anyone who can fix the NURBS backend right now.