whitequark/solvespace #178: Length related constraints for Bezier curve

Symbian9, 2016-05-20:
Think, it must be one of priority task.
* Length of curve (similar to line distance)
* Length Ratio (between curve and other curve, line, arc or circle)
  * https://github.com/whitequark/solvespace/issues/60
* Length Difference (between curve and other curve, line, arc or circle)
* Equal Length (between curve and other curve, line, arc or circle)

How calculate length of curve:
* https://github.com/whitequark/solvespace/issues/163#issuecomment-220527646

Related:
* https://github.com/whitequark/solvespace/issues/52
* https://github.com/whitequark/solvespace/issues/163

Evil-Spirit, 2016-05-20:
We have to make ability to introduce new paramters and use it in expression formulas. There is no need to copy-paste for length ratios, length difference and so on. We just let users do what they want: https://youtu.be/jA2bjEPhQkM?t=1m40s

Symbian9, 2016-05-20:
@Evil-Spirit, think not all users like use formulas, and more like traditional menu option entry.

Evil-Spirit, 2016-05-20:
@Symbian9, there is no point to implement features which produces сombinatorial explosion, if there is short and elegant way to avoid this.

Symbian9, 2016-05-20:
Anyway, I want know point of view for this future from @whitequark and @jwesthues

whitequark, 2016-05-20:
These constraints make very little sense, because there are so many parameters for the solver to adjust that will change length of a Bezier curve. Sure, it does *something* to make it the desired length, but is that the result you expect? Likely not.

And besides, I cannot imagine applications for this constraint anyway.

closed 2016-05-20 by whitequark

Evil-Spirit, 2016-05-20:
@whitequark, actually, we can implement length constraint for rational beziers, and this is quite simple equations. But really, I also can not imagine the applications.

whitequark, 2016-05-20:
Yes, we can do that, and the solver will to move control points around, but when exactly is that sort of optimization desirable? It's going to produce completely random movement unless you add a lot more constraints on control points.

jwesthues, 2016-05-21:
I don't think there's any useful closed form, so algebraic constraints probably wouldn't do this, unless they implemented a special "arc length by numerical integration" function. In any case, this seems obviously feasible, and not obviously useful.