whitequark/solvespace #180: U+0454 character bug inside "unicode.lff"

Symbian9, 2016-05-20:
I found that inside character "є" there is mistake (now internal font of SolveSpace have not correct character of it)
[image]

Evil-Spirit, 2016-05-20:
Pls, look at LibreCAD.

Symbian9, 2016-05-20:
@Evil-Spirit, I was an author of patches (and I make few ukrainian characters specially for SolveSpace)
* https://github.com/whitequark/solvespace/issues/94
* https://github.com/LibreCAD/LibreCAD/issues/690
When make this characters  I use LibreCAD and in LibreCAD it shown normally. 
Now I dont know why its happen with only this character in SolveSpace.
So, think most closer look of this character in LibreCAD's fonts is inside "isofont.lff"

I sew that on LibreCAD you find some bug, but is it fixed?
In latest SolveSpace 2.1rc1 this character still broken

whitequark, 2016-05-20:
@Symbian9 I applied your patch exactly...

whitequark, 2016-05-20:
@Symbian9  Oh, actually this is because your є character is broken. With the new, more rigorous font parser it crashes SolveSpace (this is not yet released as any nightly or so).

Specifically:
```
[0454] є
4,0;1.5,0;-0,1.5,A-0.41;0,4.5;1.5,6,A-0.41;46←
2.5,3;0,3
```
To the left of the arrow should be an `X,Y` coordinate, not this.

Symbian9, 2016-05-20:
@whitequark, so please, try recreate this character or try get it from "cyrillic_ii.lff" (this font have same character that look exactly as it must look in "unicode.lff")
* https://github.com/LibreCAD/LibreCAD/blob/master/librecad/support/fonts/cyrillic_ii.lff

whitequark, 2016-05-20:
Fixed in 9c08398.

closed 2016-05-20 by whitequark

Symbian9, 2016-05-20:
@whitequark, thank you.