whitequark/solvespace #181: Inserting some character in comment text input field bug

Symbian9, 2016-05-20:
Step to reproduce:
* open SolveSpace
* create circle and apply diametre
* go to styles and set inside "s006-#def-constraint" text  style height to 100 pixels
* add comment to this sketch and double click for edit it
* insert <code>И</code>(U+0418) or  <code>и</code> (U+0438) character, maybe some other too
* put "Enter" key for approve text

RESULT: main window automatically enlarged and rendering proportion have mistake (circle would be shown as ellips).

In terminal reported this:
<code>$ solvespace
home/travis/build/solvespace/solvespace/src/gtk/gtkmain.cpp: 339:  glGetError() == 0x501 invalid value</code>

whitequark, 2016-05-20:
Don't do that.

whitequark, 2016-05-20:
It's a problem with GTK that has no easy solution and I'm not interested in spending time on it.

Symbian9, 2016-05-20:
@whitequark, but its a real bug! What a problem here? 

So, maybe, need report to GTK dev team?

Symbian9, 2016-05-20:
Also, can approve that its this bug depend from text size of comment for all characters (now reproduce it using only english characters with text size 100 pixel)

whitequark, 2016-05-20:
Fixed in 5592ad6.

closed 2016-05-20 by whitequark

Symbian9, 2016-05-21:
@whitequark, thank you.