whitequark/solvespace #183: Cannot reset text size in standart styles using factory settings

Symbian9, 2016-05-22:
Now when change size of text inside standart styles (that have #def in name) and after that try use "load factory defaults" command text size still look as defined by user and NOT as it must be (default text size in SolveSpace is 11.5 px).

Symbian9, 2016-05-22:
Additional info to reproduce:
* run SolveSpace
* change text height inside <code>s006-#def-constraint</code> style to 60 px (for example)
* save this file
* close SolveSpace
* reopen SolveSpace
* default text for comments now look as 60 px
* change it again to 11.5 px
* add any comment text
* try "load factory defaults"

Think, SolveSpace least saved style as factory defaults

UPD: <b>this bug also work for colors and other style parametres too</b>

Evil-Spirit, 2016-05-26:
Ok ill do it

Evil-Spirit, 2016-05-26:
Fixed

whitequark, 2016-05-26:
Fixed in 73d82a6.

closed 2016-05-26 by whitequark