whitequark/solvespace #185: FreeBSD: fails to link because of execinfo

melak, 2016-05-26:
FreeBSD 10.3/amd64

Compiled with
```bash
cmake \
        -DGUI="gtk3" \
        -DCMAKE_CXX_FLAGS="-g -ggdb3" \
        ../solvespace/

make VERBOSE=1
```
Fails to link because ```execinfo``` not being present among the link libraries.

```bash
[ 93%] Linking CXX executable solvespace
cd /home/ice/w/solvespace/bin/src && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/solvespace.dir/link.txt --verbose=1
/usr/bin/CC   -g -ggdb3 -stdlib=libc++ -Wall -Wextra -Wno-unused-parameter -Wfloat-conversion -Werror=switch  -Wl,--as-needed  CMakeFiles/solvespace.dir/util.cpp.o CMakeFiles/solvespace.dir/entity.cpp.o CMakeFiles/solvespace.dir/expr.cpp.o CMakeFiles/solvespace.dir/constraint.cpp.o CMakeFiles/solvespace.dir/constrainteq.cpp.o CMakeFiles/solvespace.dir/system.cpp.o CMakeFiles/solvespace.dir/platform/unixutil.cpp.o CMakeFiles/solvespace.dir/platform/gtkmain.cpp.o CMakeFiles/solvespace.dir/platform/gloffscreen.cpp.o CMakeFiles/solvespace.dir/bsp.cpp.o CMakeFiles/solvespace.dir/clipboard.cpp.o CMakeFiles/solvespace.dir/confscreen.cpp.o CMakeFiles/solvespace.dir/describescreen.cpp.o CMakeFiles/solvespace.dir/draw.cpp.o CMakeFiles/solvespace.dir/drawconstraint.cpp.o CMakeFiles/solvespace.dir/drawentity.cpp.o CMakeFiles/solvespace.dir/export.cpp.o CMakeFiles/solvespace.dir/exportstep.cpp.o CMakeFiles/solvespace.dir/exportvector.cpp.o CMakeFiles/solvespace.dir/file.cpp.o CMakeFiles/solvespace.dir/generate.cpp.o CMakeFiles/solvespace.dir/glhelper.cpp.o CMakeFiles/solvespace.dir/graphicswin.cpp.o CMakeFiles/solvespace.dir/group.cpp.o CMakeFiles/solvespace.dir/groupmesh.cpp.o CMakeFiles/solvespace.dir/importdxf.cpp.o CMakeFiles/solvespace.dir/mesh.cpp.o CMakeFiles/solvespace.dir/modify.cpp.o CMakeFiles/solvespace.dir/mouse.cpp.o CMakeFiles/solvespace.dir/polygon.cpp.o CMakeFiles/solvespace.dir/resource.cpp.o CMakeFiles/solvespace.dir/request.cpp.o CMakeFiles/solvespace.dir/solvespace.cpp.o CMakeFiles/solvespace.dir/style.cpp.o CMakeFiles/solvespace.dir/textscreens.cpp.o CMakeFiles/solvespace.dir/textwin.cpp.o CMakeFiles/solvespace.dir/toolbar.cpp.o CMakeFiles/solvespace.dir/ttf.cpp.o CMakeFiles/solvespace.dir/undoredo.cpp.o CMakeFiles/solvespace.dir/view.cpp.o CMakeFiles/solvespace.dir/srf/boolean.cpp.o CMakeFiles/solvespace.dir/srf/curve.cpp.o CMakeFiles/solvespace.dir/srf/merge.cpp.o CMakeFiles/solvespace.dir/srf/ratpoly.cpp.o CMakeFiles/solvespace.dir/srf/raycast.cpp.o CMakeFiles/solvespace.dir/srf/surface.cpp.o CMakeFiles/solvespace.dir/srf/surfinter.cpp.o CMakeFiles/solvespace.dir/srf/triangulate.cpp.o  -o solvespace  -L/usr/local/lib  ../extlib/libdxfrw/libdxfrw.a /usr/local/lib/libGLU.so /usr/local/lib/libGL.so -lpng16 -lz -lfreetype -lgtkmm-3.0 -latkmm-1.6 -lgdkmm-3.0 -lgiomm-2.4 -lgtk-3 -lgdk-3 -latk-1.0 -lcairo-gobject -lgio-2.0 -lgdk_pixbuf-2.0 -lpangomm-1.4 -lglibmm-2.4 -lcairomm-1.0 -lsigc-2.0 -lpangocairo-1.0 -lpango-1.0 -lgobject-2.0 -lglib-2.0 -lintl -lcairo -lpthread -lX11 -ljson-c -lfontconfig -lfreetype -lGLEW -lGLU -lGL -lgtkmm-3.0 -latkmm-1.6 -lgdkmm-3.0 -lgiomm-2.4 -lgtk-3 -lgdk-3 -latk-1.0 -lcairo-gobject -lgio-2.0 -lgdk_pixbuf-2.0 -lpangomm-1.4 -lglibmm-2.4 -lcairomm-1.0 -lsigc-2.0 -lpangocairo-1.0 -lpango-1.0 -lgobject-2.0 -lglib-2.0 -lintl -lcairo -lpthread -lX11 -ljson-c -lfontconfig -lGLEW -lGLU -lGL -Wl,-rpath,/usr/local/lib: 
CMakeFiles/solvespace.dir/platform/unixutil.cpp.o: In function `SolveSpace::assert_failure(char const*, unsigned int, char const*, char const*, char const*)':
/home/ice/w/solvespace/solvespace/src/platform/unixutil.cpp:35: undefined reference to `backtrace'
/home/ice/w/solvespace/solvespace/src/platform/unixutil.cpp:36: undefined reference to `backtrace_symbols'
c++: error: linker command failed with exit code 1 (use -v to see invocation)
*** Error code 1
```

After having a quick experiment with
```diff
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index 9ec43b9..22da2b2 100644
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -120,11 +120,16 @@ elseif(HAVE_GTK)
         platform/gtkmain.cpp
         platform/gloffscreen.cpp)
 
+    if(${CMAKE_HOST_SYSTEM_NAME} STREQUAL "FreeBSD")
+        find_library(execinfo_LIBRARY execinfo /usr/lib /usr/local/lib)
+    endif()
+
     set(platform_LIBRARIES
         ${GTKMM_LIBRARIES}
         ${JSONC_LIBRARIES}
         ${FONTCONFIG_LIBRARIES}
-        ${GLEW_LIBRARIES})
+        ${GLEW_LIBRARIES}
+        ${execinfo_LIBRARY})
 endif()
 
 # solvespace executable
```
solvespace itself builds successfully and runs too, however CDemo still doesn't link:

```bash
[100%] Linking C executable CDemo
cd /home/ice/w/solvespace/bin/exposed && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/CDemo.dir/link.txt --verbose=1
/usr/bin/cc   -Wall -Wextra -Wno-unused-parameter -Wfloat-conversion -Werror=switch  -Wl,--as-needed  CMakeFiles/CDemo.dir/CDemo.c.o  -o CDemo  ../src/libslvs.so.3.0 -Wl,-rpath,/home/ice/w/solvespace/bin/src 
../src/libslvs.so.3.0: undefined reference to `backtrace_symbols'
../src/libslvs.so.3.0: undefined reference to `backtrace'
cc: error: linker command failed with exit code 1 (use -v to see invocation)
*** Error code 1
```
That makes me think the above diff is not enough or just plain not the right place to wedge ```libexecinfo``` in.

whitequark, 2016-07-19:
I've pushed a nicer fix (using CMake's FindBacktrace and also compatible with OpenBSD) in master. Please test and reopen if it doesn't work for you somehow.

closed 2016-07-19 by whitequark

melak, 2016-07-26:
Works excellent, cheers!