whitequark/solvespace #190: Rewrite text window screens using templates

whitequark, 2016-06-01:
Necessary for internationalization.

whitequark, 2016-11-28:
@Evil-Spirit I want to rewrite the rendering code in text window. How do you think is it better to implement drawing? By storing one string with \n and color markers and such? Or some more complicated data structure, like a 2d array, or a tree or something?

Evil-Spirit, 2016-11-28:
@whitequark, If think 2d array is good (as it done now).

Evil-Spirit, 2016-11-28:
@whitequark 
For saving memory it can be good to use something like style;
```
struct TWCell {
    uint32_t symbol;
    TWStyle *style;
    TWLink *link;
}
```

whitequark, 2016-11-28:
> For saving memory it can be good to use something like style;

this is really awful for cache.

Evil-Spirit, 2016-11-29:
@whitequark,
This doesn't matter since we have small amount of elements (rows and cols for drawing is calculating with corresponding with TextWindow content and window size)

whitequark, 2016-11-29:
then why bother allocating them at all? just render on the fly.

Evil-Spirit, 2016-11-29:
@whitequark 
I think this is because we have to find what element we are hovering and this is just simple in case of recangular grid, but is case of plain string we have to iterate over it since we can't just direcly map row,col.

whitequark, 2016-11-29:
@Evil-Spirit Yes. The idea was to store row offsets into string, and do binary search for those first

Evil-Spirit, 2016-11-29:
@whitequark,
I think it is enough to create iteartor for string which can iterate over elements and return element row/col or detailed info on demand (color, link, etc). For hovering it will be just linear search in cache line, and for drawing it is just iterating.

Evil-Spirit, 2016-11-29:
@whitequark, And yes, we can store at least row offsets for faster search. But I think this will be fast enough even without it.

Evil-Spirit, 2016-11-29:
@whitequark,
I think the main ideas of refactoring of text window should be:
* ability to use templates from files to prepare for translation
* ability to open multiple tw (I need at least two - for groups and for dynamic info about solve result or selection)
* simple representation of tw strings
* ability to dock tw into main window or to each other

whitequark, 2016-11-29:
> ability to open multiple tw (I need at least two - for groups and for dynamic info about solve result or selection)

Bad idea. Multiple TW won't work on android or web.

> ability to dock tw into main window or to each other

This is actually done in single-window branch already, but misses a splitter and scrollbar UI.

Evil-Spirit, 2016-11-29:
@whitequark, we can perform docking by combining into same opengl viewport, so this will work fine.

whitequark, 2016-11-29:
> @whitequark, we can perform docking by combining into same opengl viewport, so this will work fine.

That's actually the easy part, the hard part is handling input, resizing, laying out etc

Evil-Spirit, 2016-11-29:
@whitequark, also not a problem. I have been implemeting some gui things https://github.com/Evil-Spirit/Nutmeg/tree/master/CoreSrc/Gui

closed 2018-04-03 by whitequark