whitequark/solvespace #194: Parametric sketches

whitequark, 2016-06-01:


Evil-Spirit, 2016-06-01:
For this, we sould introduce parameter table for the sketch.
For each parameter:
* value
* name
* "free" or "variable" - ability to solve it through solver (otherwise, this is just constant).
* min, max (for ranged variable parameters)

For defining, we can use parameter table, or just type new name inside text input field of constraint:

[image]

jasonk, 2016-06-25:
It would be also be awesome if you could have multiple sets of parameters and be able to switch between them.  So, for example, in addition to the regular settings, I could switch to a set of parameters that changes the spacing between assemblies, so I could easily switch back and forth between a normal assembled view and an exploded view.

whitequark, 2016-06-26:
Yeah, that's planned. Though exploded view is a separate feature.

ryannining, 2016-07-05:
i change some code, using `comment` on constraint to keep the expresion editable.

And then i use `constraint comment` to introduce named variable (parsing `=` and save to MAP). 

Just the problem is, when file is saved then loaded, the named variable is not loaded, need to edit comment and immidiately press return to process the expression.

![Imgur]
(http://i.imgur.com/bDDEuYL.png))

whitequark, 2016-10-14:
Migrated to solvespace/solvespace

closed 2016-10-14 by whitequark