whitequark/solvespace #195: Signed distance, angle, etc constraints

whitequark, 2016-06-01:
This is a prerequisite for #194; without this sketches can wildly change orientation when changing just constraint values.

Evil-Spirit, 2016-06-01:
I've found what this can be called "preserving chirality"
https://en.wikipedia.org/wiki/Chirality

we can do it for:
* PROJ_PT_DISTANCE
* PT_LINE_DISTANCE
* PT_PLANE_DISTANCE
* PT_FACE_DISTANCE
* SYMMETRIC
* SYMMETRIC_HORIZ
* SYMMETRIC_VERT
* SYMMETRIC_LINE
* HORIZONTAL
* VERTICAL
* PERPENDICULAR
* ANGLE
* ARC_LINE_TANGENT
* CUBIC_LINE_TANGENT
* CURVE_CURVE_TANGENT
* PARALLEL

Evil-Spirit, 2016-06-01:
For implementing this we should introduce for each constraint:
* "option" (pos/neg, parallel/antiparallel)
* "preserve chirality" (ability to trun off "sign" considering and turn back to the old behaviour)

whitequark, 2016-06-01:
This will not be done with old file format anyway.

Evil-Spirit, 2016-06-01:
@whitequark, ofc

whitequark, 2016-10-14:
Migrated to solvespace/solvespace

closed 2016-10-14 by whitequark