whitequark/solvespace #198: OpenGL core profile renderer

whitequark, 2016-06-01:
This allows us to render on Android, Wayland, or WebGL; and elsewhere, to batch requests, which will vastly improve performance.

Evil-Spirit, 2016-06-03:
Ok. Can i do this?

whitequark, 2016-06-03:
Sure. There should be everything in place already.

Evil-Spirit, 2016-06-17:
[image]

whitequark, 2016-06-18:
Very nice

Evil-Spirit, 2016-06-21:
[image]

Evil-Spirit, 2016-06-21:
There is still some problem with precision if we apply big values for stipple scale (because of using just 8bits). We can fix it by using F32 on desctops, or packing floats to rgba.
[image]

Evil-Spirit, 2016-06-21:
[image]

Evil-Spirit, 2016-06-23:
Stipple phase stiching for entities
[image]

Evil-Spirit, 2016-06-23:
Stitched all the stipples and removed
[image]
 duplicates using the PolylineBuilder.

Evil-Spirit, 2016-06-23:
[image]

Evil-Spirit, 2016-06-24:
Full-shader outline rendering.
visiblity compuded inside VS, used "discard" in FS.

[image]

Visibility test supressed:
[image]

Evil-Spirit, 2016-06-24:
https://github.com/Evil-Spirit/solvespace/commit/e190e7dc91a66545e09cd4ad72a6cf8c6adfc7a8

whitequark, 2016-06-24:
great!

Evil-Spirit, 2016-06-26:
Draw emphasized with outlines in the same pass.

* less generation: there is no need to generate emphasized edges separately.
* less geometry drawing
* less geometry overdraw (we can draw only emphasized WITHOUT outlines for two-pass wide outlines rendering)
* ability to stitch stipple phase inside the generation stage to produce high-quality result

[image]

Evil-Spirit, 2016-06-26:
@whitequark , what do you think about vbo chaching approach like here https://github.com/Evil-Spirit/solvespace/commit/0d8cd6af0c1be98fbcefb9289c287b76fd6e94e0

Evil-Spirit, 2016-06-26:
Screwed up some perfomance and assembly abilities.
Now we can do only 25 bearings in good precision (laptop i7-3610qm + GT 650m)

[image]

[image]

Evil-Spirit, 2016-06-30:
[image]

Evil-Spirit, 2016-07-28:
All aspects of OpenGL2-shader renderer implementation is done. 
[image]

Evil-Spirit, 2016-07-29:
@whitequark, rebased onto master https://github.com/Evil-Spirit/solvespace/tree/opengl2

Evil-Spirit, 2016-07-29:
Still in progress, but you can review.
1. We need to eliminate Canvas member variables and replace it with member functions.
2. We need to create and hold canvas for our windows.
3. I don't see any reason to split this class into two renderers - all what we draw using trivial primitives goes dynamic way, all big objects are cached using pointers and versions(timestamps).
4. Point is like line, so we have use hStroke for it. If we need hFill, we can use drawQuad.
5. We can make phase considering for DoStippledLine and support stipple stitching.
6. We must abstract out this std::crono shit to double getSeconds();

whitequark, 2016-08-02:
> 3. I don't see any reason to split this opengl2 class into two renderers - all what we draw using trivial primitives goes dynamic way, all big objects are cached using pointers and versions(timestamps).

Well, this "version" idea is a bad one. It places too much responsibility into the renderer, and it violates the abstraction of SMesh, SEdgeList, etc, which should be just dumb storage.

whitequark, 2016-08-02:
And why is there any immediate mode code at all? That's just unnecessary complexity.

Evil-Spirit, 2016-08-02:
> And why is there any immediate mode code at all? That's just unnecessary complexity.
What the problem with it?

Evil-Spirit, 2016-08-02:
> Well, this "version" idea is a bad one. It places too much responsibility into the renderer, and it violates the abstraction of SMesh, SEdgeList, etc, which should be just dumb storage.

Do you have any suggestions? Is better to use approaches with handles? hMesh, hOutlines and so on? We can have somtihng like hMesh MakeMesh(const Mesh &m) functions, but this needs to change canvas interface and change code to store and generate.

whitequark, 2016-08-02:
I already explained it twice. The renderer would not do any caching decisions at all. It would provide an `OpenGL2Batch : public Canvas`, which would be populated by whatever wants to render stuff, and which would remain stored forever (or until destroyed). The decision to invalidate the cache would rest on the CAD core, not on the renderer.

Evil-Spirit, 2016-08-02:
Ok, then I have to change Canvas interface.
`
hMesh Get/GenMesh(const Mesh &mesh);
void DrawMesh(hMesh mesh);
`
Is this what you mean?

Evil-Spirit, 2016-08-02:
Or you mean approach you have used with Pixmap?
```
void DrawMesh(const Mesh &mesh);
void InvalidateMesh(const Mesh &mesh);
```

whitequark, 2016-08-02:
No. None of the above. There would be *no* caching inside the renderer. No InvalidateMesh needed. (Actually, there is no need for InvalidatePixmap *between* renders either, the only reason InvalidatePixmap exists is drawing bitmap font; it's not and will not be used for anything else.)

A Batch is populated once, and thrown away entirely once it's no longer needed. So there is no caching going on in the renderer, and no cache invalidation.

Evil-Spirit, 2016-08-02:
Why? We no need to fill video memory buffers every draw call.

whitequark, 2016-08-02:
That won't happen. Persistent stuff like entities would simply be put into a Batch once and then reused until the next regeneration. It would be a single DIP call every frame until then, no refilling.

whitequark, 2016-08-02:
As for the benefits, apart from having a simpler renderer, there are others. For example you don't actually need to keep the mesh in RAM, you can just put it in the video RAM once and then destroy.

Evil-Spirit, 2016-08-02:
We have to manipulate meshes separately when we will want to get rid of assembly mesh regeneration. We can't do such mess now and lock ability to draw the same cached mesh many times with the different transformations.

whitequark, 2016-08-02:
That is also something that's easier with the Batch interface. Just create a Batch per sketch, then draw it using different projection matrix. Since a Batch doesn't have a working GetCamera() (it just asserts) then anything drawn into a batch is automatically guaranteed to be independent of camera position, and so it is safe to draw with any projection matrix.

Evil-Spirit, 2016-08-02:
So why you archtecturaly don't distingush Canvas from GraphicsAPI? We have OpenGL1 API, OpenGL2 API, and OpenGL2Batch - the array of what we wanted to draw using OpenGL2 API/

whitequark, 2016-08-02:
Yes, there would be a separate OpenGL2Renderer (not a canvas) and OpenGL2Batch (canvas). I haven't designed the final interface yet, changes will have to be made for OpenGL2.

whitequark, 2016-10-23:
@jwesthues This branch is nearing completion. It is a hard requirement for mobile/web GL as well as the new Linux graphics server (Wayland) and it provides significant performance improvement elsewhere.

Do you have any particular opinion on the support for OpenGL 1? The current plan is as follows:
  * drop OpenGL 1 renderer completely;
  * use OpenGL 2 core profile on macOS directly, since there is no hardware that still receives OS updates that lacks support for that GL version, and the software fallback for e.g. VMs is only a little slower than using GL1;
  * use OpenGL 2 core profile on Linux directly, since the hardware and driver support are widely available; there is still hardware that doesn't support GL 2, like some netbooks, and software fallback is available through LLVMpipe, but hardware with such poor integrated graphics is generally already too slow to usefully render in software; and so anyone with such hardware can continue using SolveSpace 2;
  * use ANGLE with the D3D9 backend on Windows; this avoids the common issue where no OpenGL drivers are installed on the target system; not all hardware supports D3D9, but WARP provides a fast compatibility layer since Vista; XP systems are probably out of luck, but given that they are unsupported for a long time and are extremely insecure to run, I do not feel that we have a strong obligation to continue Windows XP support; anyone who still uses Windows XP can continue using SolveSpace 2.

jwesthues, 2016-10-23:
I haven't recently compared options for this, but your plan seems generally reasonable. I'd agree that if Microsoft has dropped support for XP, then it's reasonable for us to as well.

Is the reason not to use ANGLE outside Windows speed, or just general unnecessariness?

whitequark, 2016-10-23:
> Is the reason not to use ANGLE outside Windows speed, or just general unnecessariness?

It's impossible to meaningfully use ANGLE as it only supports the core profile GL (desktop and ES) and D3D9/D3D11 as backends. But if the system already provides the core profile GL APIs, then what's the point of using ANGLE? Chrome uses it to achieve consistent behavior across platforms wrt error reporting in user-provided shaders in WebGL, but we don't need anything like that, and it's one less component to integrate and ship. (Mesa on Linux does provide the D3D API, but they have feature parity with GL APIs, and that doesn't work for most discrete video cards, so it's even more pointless.)

ruevs, 2016-10-25:
Dropping support for XP will make me sad... But I guess by now I am in the 10%
https://www.netmarketshare.com/operating-system-market-share.aspx?qprid=10&qpcustomd=0

ruevs, 2016-10-25:
Better drop support for all versions of Mac OS together you will lose fewer potential users.

whitequark, 2016-10-25:
> Better drop support for all versions of Mac OS together you will lose fewer potential users.

... and also all Linux versions, no? But in any case, that's only correct if people who are most likely to use a CAD program are evenly distributed across operating systems, which is of course not the case.

whitequark, 2016-10-25:
@ruevs FWIW I've just looked a bit closer at it and it *seems* that ANGLE will still work with XP, but I make no particular promises at this time.

ruevs, 2016-10-25:
In fact Solvespace is one of the few pieces of software that will convince me to install 7 on real hardware (already have one in a virtual machine) if it dropped XP support - that is how much I like it.
And supporting XP is only worth it if it is not a trade off with portability/clean architecture/speed.

whitequark, 2016-11-16:
@ruevs Here's how it turned out:
  * ANGLE does not easily build targeting Windows XP as the shader compiler is missing from the v120_xp toolkit for some reason.
  * VirtualBox's 3D acceleration is completely broken. (It is broken even with SolveSpace 2.)
  * Windows 7 seems to lack software fallback for D3D9.
  * Windows 10 has software fallback for D3D9, and it works reliably though slow (remarkably it's slower than the software fallback in macOS, which is particularly silly because macOS ought to never use said fallback on Apple hardware, whereas it's not true for Windows).

In conclusion it will likely be unviable to run the released SolveSpace 3 on anything older than Vista with real D3D9+ hardware. (IIRC, Vista requires D3D9+ hardware to install anyway).

For now I have left the OpenGL 1 renderer in the tree, and you can make a build of SolveSpace that will run on XP, by using Visual Studio 2013 and configuring it as `cmake -G"Visual Studio 12" -T v120_xp -DOPENGL=1`, or by using mingw and configuring it as `cmake -DOPENGL=1`.

whitequark, 2016-11-16:
Issue migrated to https://github.com/solvespace/solvespace/issues/102, let's continue any discussion there.

closed 2016-11-16 by whitequark