whitequark/solvespace #202: SLVF - SolveSpace font file format

Symbian9, 2016-06-09:
I think it would be cool if SolveSpace can be fully independent from external fonts using own font format, and user can make own symbol set directly in SolveSpace.
(as I imaginate this would be similar how LibreCAD managed own fonts)

Principles:
* SLVF - is normal SLVS file (with only *.slvf extension), where one group, named as only 4 character codes of UTF8 symbols, include sketch of text symbol. For example, <code>g002-0040</code> must include only sketch of @ symbol.
* SLVF font can be added to SLVS similar as now parts added using "Assemble" operation
* name of font is same as name of SLVF file (without extension)
* Different characters inside one font can have different colors

P.S.: this is proof of concept, so later I will update this issue with more specific info

Evil-Spirit, 2016-06-09:
I think there is no point to make one more font format. SolveSpace is not so popular, so there is no enough people who can support and contribute fonts. LibreCad font format covers all the functions we need.

Symbian9, 2016-06-09:
Currently only one LFF font SolveSpace can use, and this font still buggy in some characters. For fixing it need additionally install LibreCAD(+huge amount of Qt libs...)

Now user cant select other LibreCAD font in SolveSpace if he is not an programmer and don't recompile it itself.

And there are no any other way how use another plotter (single line) fonts in SolveSpace. But for real production we need at least ISO (International & EU standarts) and GOST  (exUSSR standarts) fonts inbuilt.

I very like this easy way of 2D drawing process in SolveSpace and hope that SLVF fonts would be implemented anyway, so, please, DO NOT CLOSE THIS ISSUE now. I will try work on it soon, firstly trying make detailed process infographics for this option.

Evil-Spirit, 2016-06-09:
> Now user cant select other LibreCAD font in SolveSpace if he is not an programmer and don't recompile it itself.

I think we could change this behaviour (there will be one embedded default font and ability to load different lff files)

Symbian9, 2016-06-09:
@Evil-Spirit, yes, importing LFF would be good start.

But have own font format, that can be easy edited directly in SolveSpace is good thing.
Maybe, its possible add LFF editing futures for SolveSpace too?

Evil-Spirit, 2016-06-09:
Maybe it's good idea to load and save lff the same way as LibreCAD do. But it is not priority for now. The development process is like spiral, so we could return this task in the future.

whitequark, 2016-06-09:
@Symbian9 No. There are just three or four useful vector fonts in existence and spending time writing, debugging and maintaining code in SolveSpace just so that you could avoid installing Qt libraries is an incredible waste of time. I will not implement this.

Further, there are plenty of existing font formats and font authoring systems apart from `.lff`, and you are welcome to implement a converter from these formats to `.lff`, which is trivial.

closed 2016-06-09 by whitequark

whitequark, 2016-06-09:
Oh and by the way, librecad already provides a ttf2lff converter, so you can use any ttf font you want, or any ttf authoring tools you want.

whitequark, 2016-06-11:
> Now user cant select other LibreCAD font in SolveSpace if he is not an programmer and don't recompile it itself.

@Symbian9 This is no longer true for the latest master. You can place any .lff font in `/usr/share/solvespace` and it will be picked up. Note that it is MANDATORY that the characters are sorted in the order of their Unicode codepoint number.