whitequark/solvespace #208: Weird problem with midpoint constraints

whitequark, 2016-06-22:
See [top.zip](https://github.com/whitequark/solvespace/files/328813/top.zip). It's unsatisfied. I see no reason it should be.

whitequark, 2016-06-23:
Another problem of the same class: 
[top.zip](https://github.com/whitequark/solvespace/files/329983/top.zip)

whitequark, 2016-06-23:
Looks like in 2nd case the culprit is a duplicated vertical constraint. This error should be handled better.

whitequark, 2016-06-23:
In 2nd case this is a duplicated vertical and horizontal constraint. We should show an error when those constraints are added, not a much more confusing one afterwards.

Evil-Spirit, 2016-06-23:
the problem is what we don't put all the equations (for dof analysis) into Jacobian, so we have no ability to alert it like "overconstrained"

whitequark, 2016-06-23:
How can we fix this?

Evil-Spirit, 2016-06-23:
we just can lookup all the equations and try to find. or we can try to add all the equations to Jacobian. Test and reporting will run in natural way (this is only for the first time when we add constraint).

whitequark, 2016-06-23:
Let's try the latter one (but be careful about slowing down, it is already quite slow)

Evil-Spirit, 2016-06-23:
It's not slow because of solver I think.

Evil-Spirit, 2016-06-23:
hm... it looks like this is not the fix(don't solve by substitution)

Evil-Spirit, 2016-06-23:
Oh, no, i've missed and commented not related line inside FindWhichToRemoveToFixJacobian(). When we comment SolveBySubstitution() inside Solve() we get the right "redundant" message.

Evil-Spirit, 2016-10-22:
@whitequark,
I've pushed the prototype. I propose not adding the constraints which causes errors like overdefining the system or not converging from the start.

whitequark, 2016-10-23:
But this leaves the sketch in a broken state afterwards, if solving has moved the stuff around.

Evil-Spirit, 2016-10-23:
@whitequark, did you mean changed? if rudundant constraints solve sketch and copy it's parameters, it will be changed. So it just can be fixed by not coping parameters when test-solving.

Evil-Spirit, 2016-10-23:
fixed here https://github.com/Evil-Spirit/solvespace/commit/d7a80b44df3c749d2fb9dca3682236dfd0b2b24b

whitequark, 2016-11-02:
Actually, no, it is not fixed. Try this:
[image]
then create a vertical constraint on hypotenuse.

Evil-Spirit, 2016-11-02:
@whitequark, what's wrong with it?

whitequark, 2016-11-03:
@Evil-Spirit Sketch becomes broken, even though the constraint is not actually added.

Evil-Spirit, 2016-11-03:
@whitequark, Strange behaviour, can you attach screenshot? It seems I don't have such problem.

Evil-Spirit, 2016-11-03:
may be you are testing without this commit? https://github.com/Evil-Spirit/solvespace/commit/d7a80b44df3c749d2fb9dca3682236dfd0b2b24b

whitequark, 2016-11-03:
The original problem (with that commit) is as follows:
[image]

Now I am thinking, it's actually good; it should show the "unsolvable constraints" thing in the text window to demonstrate that there is a conflict. But the text in text window should be replaced with something else, like "cannot add this constraint because of a conflict with one of these:" or something.

Evil-Spirit, 2016-11-03:
@whitequark,
Yes, the text in messagebox must be rewritten.

Evil-Spirit, 2016-11-06:
@whitequark, more improvements
https://github.com/Evil-Spirit/solvespace/commit/9e6d030458bea9d3c1adafb85780ca12260daca5
https://github.com/Evil-Spirit/solvespace/commit/e11fa8fb82140daee90111198dd30c9729560f7c

Evil-Spirit, 2016-11-27:
@whitequark 
you can merge this

whitequark, 2016-11-27:
I don't like the UI very much. We used to have red screen when nonsensical constraints are inserted and when a dof is removed. Now we have red screen when a dof is removed, and a popup when a nonsensical constraint is inserted. And this interrupts workflow even more.

I need to think about fixing this best.

Evil-Spirit, 2016-11-28:
@whitequark 
I only can say what this is much better for user not allowing add unsolveable and redundant constraints. Anyway, the standart way to fix it just press ctrl+z. This breaks workflow, not interrupt since both of them interrupts workflow some way. I agree what user must see what's wrong with constraints, can select and decide what to do. But for me (and for some other users) is just good to auto-remove new defective constraint immediately. I think this can be an option at least. But what do you think about auto-marking constraint with reference if it makes sketch redundant? This is usefult and not breaking anything

whitequark, 2016-11-28:
> I only can say what this is much better for user not allowing add unsolveable and redundant constraints.

Yes. I think we need this functionality in some form, just not yet sure which.

> But what do you think about auto-marking constraint with reference if it makes sketch redundant? This is usefult and not breaking anything

It is but it doesn't work for most constraints.

Evil-Spirit, 2016-11-28:
> Yes. I think we need this functionality in some form, just not yet sure which.

This can be an option "Disable red-screen" which also changes behaviour of dragging (just reverting parameters when sketch becomes unsolveable). I actually imlemented such workflow and there is no problems with this.

> It is but it doesn't work for most constraints.

Yes, this can be only applicable for constraints with dimension label. Actually, this is just standart approach for CAD applications.

Evil-Spirit, 2016-12-10:
@whitequark,
What we are going to do with?
I think we can introduce the flag : dofDirty and do the full dof check without substitution if this flag set. This  flag must be set when we add new constraint and after loading (to analyse old files for problems)
What do you think about this?

whitequark, 2016-12-10:
@Evil-Spirit sounds good

Evil-Spirit, 2016-12-12:
@whitequark 
pushed again
https://github.com/Evil-Spirit/solvespace-master/commit/c90a38ddcdbec505ad7bdd335849225e14169374

whitequark, 2016-12-21:
Merged.

closed 2016-12-21 by whitequark