whitequark/solvespace #209: Bizarre results with transparency

whitequark, 2016-06-22:
[top.zip](https://github.com/whitequark/solvespace/files/328827/top.zip)
[image]

whitequark, 2016-06-27:
This needs to be handled in the new renderer, I think.

Evil-Spirit, 2016-06-27:
@whitequark, it is the common problem in computer graphics. There is no simple solution in general;
the problem solutions you can see here https://habrahabr.ru/post/224003/
I don't think what we need to fight it now.

whitequark, 2016-06-27:
OK I see.

whitequark, 2016-12-05:
@Evil-Spirit  this is how it looks in new renderer:
[image]
which... is even weirder. I think this is because of depth buffer?

whitequark, 2016-12-05:
from the other side:
[image]

I feel maybe there's an easy fix somewhere. Are we drawing all the faces we should draw?

Evil-Spirit, 2016-12-05:
@whitequark, Sorting of triangles inside mesh by transparency should help (triangles with transparency should be the last)

whitequark, 2016-12-05:
@Evil-Spirit I think we can easily do that with the new renderer, because we only sort once, not every frame.

whitequark, 2016-12-05:
@Evil-Spirit I fixed it in master.

closed 2016-12-05 by whitequark