whitequark/solvespace #213: Select the entities in the current workplane first

whitequark, 2016-06-23:
This generally bears no relevance to semantics of constraints; it is only necessary to reduce confusion in the UI and make constraints slightly prettier.

Symbian9, 2016-06-23:
One more need is select nearest entity if two entities in view port look like placed in same place but on different layers

whitequark, 2016-06-27:
@Evil-Spirit For now don't implement this, just investigate how this can be done. Needs some fairly dramatic change to PickingCanvas...

Evil-Spirit, 2016-07-21:
@whitequark, please try it.
https://github.com/Evil-Spirit/solvespace/tree/canvas-selection

whitequark, 2016-07-21:
Can you split it into two commits? One changes the behavior wrt picking. This is what's required for merging. The other would mess with selection order. I don't understand these changes as is.

Evil-Spirit, 2016-07-22:
@whitequark, done https://github.com/Evil-Spirit/solvespace/commits/canvas-selection

whitequark, 2016-07-22:
Much better now. The code looks pretty good, I'll test soon.

Evil-Spirit, 2016-07-22:
@whitequark, any problems with this?

whitequark, 2016-07-23:
@Evil-Spirit Well, yeah, point-coincident constraints.

whitequark, 2016-07-23:
OK, I fixed the problems with selection and merged it. (Note, I only merged the code that brings the canvas branch where the old one was. This issue stays open for now.)

Evil-Spirit, 2016-07-24:
@whitequark, where can i look at this?

Evil-Spirit, 2016-07-24:
@whitequark, i ve checked out canvas and this seems to be not fixed. selection of points is bad.

whitequark, 2016-07-24:
master branch.

Evil-Spirit, 2016-07-24:
ok, tested it.
There is problems in way you have changed my commit.
You have eliminated the second minimum-distance selection (most important) which was implemented by sorting hoverList. It can be implemented without list, if don't like new selection rules like not hovering already selected entities or ability to iterate entities under cursor with pressed ctrl/shift/alt. But the calculating of the minimum distance for the current case is complete pointless, because we clear result of this every Pick() and actually no case when we can compare distance for the different entities with the same zIndex.

whitequark, 2016-07-24:
well, my goal was to merge the canvas branch while keeping the UI usable. that worked. once we have automated tests I'll revisit this issue and merge the rest, like with others.

whitequark, 2016-11-17:
Can you rebase this over master branch?

Evil-Spirit, 2016-11-17:
@whitequark, ofc

Evil-Spirit, 2016-11-17:
@whitequark,
https://github.com/Evil-Spirit/solvespace/commits/canvas-selection-over-master
But I totally don't remember what this exactly different except of selection by order.
This can be confusing in some cases, i still have no opinion about how this should work:
1. Select new entity and clear selection of an old.
2. Select new additional entity 
3. Ctrl+click for choosing 1 or 2

whitequark, 2016-11-18:
There's also https://github.com/solvespace/solvespace/pull/106... I'll think of something.

For now one thing I want is: when clicking on a point where many points are located at the same position, e.g. an extruded sketch, and the point is dragged, then the point from the earliest group should be used. This will let you drag any points of a sketch when an extrude group is active.

whitequark, 2016-11-18:
@Evil-Spirit Let's do it this way: when user clicks on the screen where there are many overlapping entities, we actually remember *two* entities:
  1. Entity in the current workplane if any, or from the last group in order otherwise. This is what goes into the "group selection" thing, and is otherwise operated on.
  2. Entity from a request. This will be used when dragging. This way, we can drag e.g. a sketch on which an extrusion group is based, even if the extrusion group is active and overlaps the sketch.

Evil-Spirit, 2016-12-10:
@whitequark,
Did you mean what we select one entity, but drag the others?

whitequark, 2016-12-10:
@Evil-Spirit correct

Evil-Spirit, 2016-12-23:
@whitequark,
https://github.com/Evil-Spirit/solvespace-master/commit/f0c60b650c689a866cbb4bea721e75736ba045fd

whitequark, 2017-01-02:
Fixed in master.

closed 2017-01-02 by whitequark