whitequark/solvespace #217: Single point-coincident constraint incompatible with itself

whitequark, 2016-06-29:
In DN100CF-fit-check.slvs. [bug.zip](https://github.com/whitequark/solvespace/files/339286/bug.zip)

Evil-Spirit, 2016-06-30:
This is known issue. We can eliminate this by solving using homogenous coordinates. This approach sucessfully tested and proven.

whitequark, 2016-06-30:
@jwesthues What's your opinion on this?

Evil-Spirit, 2016-06-30:
@whitequark, @jwesthues The reason of this behavior is disbalancing of the partial derivatives for body degrees of freedom. When we take derivatives for position parametes to write them to Jacobian mtx, we get "1.0" values. When we take derivates for rotational components, we don't get balanced derivatives (we get something like propotional to some parameter). During Newton steps the body rotated about axis very fast, but moves to target very slow until 50 steps become exceeded. The way of balancing is multipling all the coordinates by w like it performed in projective geometry.

jwesthues, 2016-07-02:
If you move the two points closer together before constraining, or if you constrain the part's orientation first, then the solver should converge.

@Evil-Spirit Can you explain in more detail what you mean by solving in homogeneous coordinates? Is the w parameter free, or constrained in some way?

Evil-Spirit, 2016-07-02:
@jwesthues, Yes, w must be constrained in some way. My math knowlege is not enough for such things, but  experiments with w = sqrt(x^2+y^2+z^2) working good for me. If you are interesting, let me know, it is better to show once than speak a lot.

Evil-Spirit, 2016-07-02:
@jwesthues, The solvespaces approach is to solve assemblies step-by-step. Because of "w" introducing in my previous project, we get ability to create parallel assemblies (solving all the equations simultaneously like 2d sketch case). We successfully build 4-cylinder engine, 24-seg Rubicks snake and so on.

jwesthues, 2016-07-02:
I have no special intuition as to why that should help, but I see no specific reason why it shouldn't. Some kind of improvement to assembly solving convergence would certainly be nice to have, and changes of this form seem unlikely to break any backward compatibility.

Evil-Spirit, 2016-07-02:
> and changes of this form seem unlikely to break any backward compatibility.
Actually not. We perform this only for writing and solving equations. We should not save w to file because we can calculate it before solving. I don't think it break backward compatibility.

whitequark, 2017-01-19:
@Evil-Spirit What about this issue? Are you working on it?

Evil-Spirit, 2017-01-19:
@whitequark, there is beta-version, but I am stuck with it. The point (0,0,0) is not common for this equations. https://github.com/Evil-Spirit/solvespace-master/commit/6ec4a0ad6d1c092f821d4453396884c5d98131da

whitequark, 2017-03-12:
@Evil-Spirit So... is there any hope of fixing it?

Evil-Spirit, 2017-03-12:
@whitequark, Beta version is making things better, but I have no theoretical base about how to solve problem with point(0,0,0). I will try to make some interation in future when my mind will be ready for this.

whitequark, 2017-03-12:
Migrated to https://github.com/solvespace/solvespace/issues/209.

closed 2017-03-12 by whitequark