whitequark/solvespace #220: solvespace as command line tools

ryannining, 2016-07-11:
Question.

Since now i have make solvespace read parameter from external text file.

Is it possible to make solvespace as command line tools ? inputing svls file and outputing stl, png preview or svg for laser cut.
Because if possible it will be great to have web based parametric part generator, user can enter parameter of the part will be generated and press update then download. It will be nice openscad replacement, because user can create content using real GUI Solvespace then share on web. Openscad require user to code instead of doing UI CAD.

Thanks.

whitequark, 2016-07-11:
No, not currently.

ryannining, 2016-07-13:
ok, i have try to hack the solvespace to compile and run on cheap vps...

![Compile, run command line, download the STL, preview in meshlab]
(http://i.imgur.com/dEp4bHf.png)

ryannining, 2016-07-13:
now trying make simple web GUI

whitequark, 2016-11-29:
Implemented in master branch.

closed 2016-11-29 by whitequark