whitequark/solvespace #224: Ray tracing performance improvements

whitequark, 2016-07-20:
Postponed until we can actually measure the impact of that

  * https://github.com/Evil-Spirit/solvespace/commit/952b83e43eb4978e8b63f53038447931122e7d28
  * https://github.com/Evil-Spirit/solvespace/commit/3125e23011bc040625ff9d1ce6fdb6e0e25829cd

Evil-Spirit, 2016-11-19:
@whitequark,
Here is some small but effective imporvement for avoiding processing of curved faces

Evil-Spirit, 2016-11-19:
This removes problem from bearing_mesh.slvs and other such things.

Evil-Spirit, 2016-11-19:
@whitequark,
Some test for big-big-big-woodworking-asm:
scaled by Entity::IsVisible = 3.63s (this still meaningful for huge amount of entities):
before: FirstIntersectionWith = 11.23s
raytrace:  FirstIntersectionWith = 2.29s
defines:  FirstIntersectionWith = 1.26s

whitequark, 2016-11-27:
Merged 29ed4de and 952b83e. Didn't merge the defines hack. I actually don't see much improvement on big-big-big-woodworking-asm from naive algorithm to raytracing, but it's already written and is a good algorithm so it's ok to merge. The defines hack seems pointless in practice. If you really want you can reimplement it with forceinline but I wouldn't bother personally, it's already fast enough.

closed 2016-11-27 by whitequark