whitequark/solvespace #225: Composing regular groups as assemblies

whitequark, 2016-07-20:
This has massive performance impact as meshes can be merged on the GPU. Postponed until after Canvas merge as this has savefile format impact

  * https://github.com/Evil-Spirit/solvespace/commit/8967d628a6de49822bbb91228a935df49f9a1d21
  * https://github.com/Evil-Spirit/solvespace/commit/65b35ca5c4741913b0dfc6bab57a7a6a0816f881
  * https://github.com/Evil-Spirit/solvespace/commit/2e39fca62f46b50ac2fd98f4d06c4bb1981810f8

whitequark, 2016-10-11:
@Evil-Spirit Commit 2e39fca62 isn't right. Right now if you do "Extrude as Difference" then "Step or Repeat" it will be subtracted multiple times. It should be that if you do "Extrude as Assemble" then "Step or Repeat" it would be assembled multiple times. There is no need in the checkbox for translate/rotate groups.

Evil-Spirit, 2016-10-11:
@whitequark, ok, we have to write function for getting "srcg" for considering step-and-transform chains.

Evil-Spirit, 2016-10-11:
I mean, when you do step-and-repeat two times, the first time all goes right way, but the second time (2-d array) we have no control for mesh operation (always performed union, with is right for union/diff, but too slow for asm)

whitequark, 2016-10-11:
Yeah, should be some function that grabs first non-translate/rotate group.

Evil-Spirit, 2016-10-11:
Ive made small fix, it works.
3d array (rotate-translate-translate)
[image]

3d array of translate
[image]

Evil-Spirit, 2016-10-11:
fix here https://github.com/Evil-Spirit/solvespace/commit/868e6d9ed91c9781be09edfc3ec1b9d8e6d93176

whitequark, 2016-10-11:
@tgirod btw, I've seen you comment about performance on HN--once this is merged your use case should become much faster. If you can share the design files I'll specifically run a profiler on them, since we're currently lacking on data for real-world benchmarks...

whitequark, 2016-10-11:
@Evil-Spirit Yeah that works.

tgirod, 2016-10-11:
@whitequark that's great news! Here is my real world project (zipped so github accepts it)

[modelisation.slvs.zip](https://github.com/whitequark/solvespace/files/522618/modelisation.slvs.zip)

whitequark, 2016-10-12:
Merged in master.

closed 2016-10-12 by whitequark

whitequark, 2016-10-13:
So I've merged the titular commit as it's useful regardless, but now I have some reservations about the supposed performance implications of this. Ideally, all assembly groups would be generated just once and then all mesh operations would be done on GPU; but now that I look closely, this doesn't seem to work if, after an assembly group, there's any other group that uses an union or difference operation, since that means a boolean operation should be made with *the entire assembly*, and so we're back to the CPU again.

So basically, we only really get the perf improvement if *every* group (perhaps past a certain point) uses the assembly operation only, and even then only until someone tries to export a thing or check for naked edges. Am I missing something?

cc @Evil-Spirit @jwesthues

whitequark, 2016-10-13:
In @tgirod's file, the top two hottest functions in the profile are...
```
  22.82%  solvespace  solvespace                     [.] SolveSpace::SSurface::MakeCopyTrimAgainst
  22.80%  solvespace  solvespace                     [.] SolveSpace::SolveSpaceUI::GroupsInOrder
```
which is quite remarkable, especially `GroupsInOrder`...

jwesthues, 2016-10-13:
@whitequark All correct, and the simple answer is not to use modes other than "assemble" in assemblies. The union and difference modes still seem useful with link/assemble, since there's otherwise no way in SolveSpace to construct a real CSG tree (vs. the usual "CSG linked list"), but will always run slower.

Ideally, we'd go further, and apply the transform (rotation and translation) to imported parts on the GPU. That would let the user drag parts in assemblies with millions of triangles around at reasonable frame rates.

That's a remarkable profile; something's certainly wrong somewhere...

whitequark, 2016-10-13:
> All correct, and the simple answer is not to use modes other than "assemble" in assemblies.

OK. The idea behind allowing "assembly" mode with extrude/lathe is so that those could then be repeated, perhaps very many times, with translate or rotate groups, e.g. think of clapboarding. Of course one could have always made a separate sketch, but those come with some overhead. I guess this still would work fine if every group has the combine mode set to "assembly", regardless of whether it's a link group or an extrude group or something, so that commit's fine.

> Ideally, we'd go further, and apply the transform (rotation and translation) to imported parts on the GPU. That would let the user drag parts in assemblies with millions of triangles around at reasonable frame rates.

Indeed, I would like to see this. Actually, it's not just triangles as entity drawing can be quite slow right now. (And entity picking is the worst offender here, that really needs to be done with a k-d tree or something.) My recent Canvas change made it possible to group all the entities into a batch and then the batch could be drawn in one glDrawElements call, which @Evil-Spirit implemented. Shaders then handle things like rotating fat lines towards the camera, stippling, and such.

So I think we can implement the assembly optimization with a heuristic: if the 1st group with a mesh is an assembly group, and some group after it isn't, then the text window would show a message, something like "this sketch could be generated faster if it consisted purely of assembly groups".

Evil-Spirit, 2016-10-13:
@tgirod 
> Here is my real world project (zipped so github accepts it)
> modelisation.slvs.zip
So big.. how do you design it with this regeneration time... ?
I thought about such application of assemble mode when made it, but don't know what someone already use it for this purpose.

tgirod, 2016-10-13:
> So big.. how do you design it with this regeneration time... ?
... painfully ;) More seriously, it got really laggy toward the end of the project, so it was ok.

What do you call "assemble mode" ?

whitequark, 2016-10-13:
@tgirod How long was regeneration in absolute terms? On my machine, it takes about 11s in debug builds and 3.5s in release builds for a full regeneration... the latter is not that bad. Also, after https://github.com/whitequark/solvespace/commit/2371068d01d0d45d660b81c451470f0effb75186 placing a new request in a sketch won't cause the entire thing to be regenerated any more, only that single request.

Evil-Spirit, 2016-10-14:
@tgirod, in new version you can change your file and specify for each extrude group "assembly" mode and your file will regenerate much faster.

jwesthues, 2016-10-14:
@tgirod "Assemble" means that instead of computing the Boolean union of the two shells, we just take all of the surfaces from both. That's always okay if the shells don't intersect, always okay for display only, not okay e.g. if you want to generate valid STL for 3d printing.

tgirod, 2016-10-14:
@whitequark When I open the file, the design appears after ~3.5 seconds, but the background is red and the UI is not responsive. It takes ~7 seconds for the UI to become responsive and for the background to turn black.

Evil-Spirit, 2016-10-23:
@whitequark 
We can also include this https://github.com/Evil-Spirit/solvespace/commit/6e9b115f7449e8583a6a8d7072e2d68d508d29c7.
For fast migration of an old files (like modelisation.slvs). Currently, it's hard to do, you should click every group, wait, click next, ... This is so annoying.

whitequark, 2016-11-17:
> That's a remarkable profile; something's certainly wrong somewhere...

@jwesthues The reason here was that GroupsInOrder was basically quadratic. That would do it...

whitequark, 2016-11-17:
@tgirod Can you release the modelisation.slvs file under some permissive CC license (e.g. CC0), so that I can include it in the solvespace repository as a performance test?

tgirod, 2016-11-17:
Of course, I have no problem with that.