whitequark/solvespace #227: Keep selected objects while change view

Symbian9, 2016-07-24:
Now if select some object(point, line, etc.) and then use change view all them automatically unselect 
For example, this issue presented after use:
* "Align View to Workplane"
* "Nearest Ortho View"
* "Nearest Isometric View"
* "Center View At Point"

But when other option from "View" submenu used all selected objects keep selected after changing view
* "Zoom In"
* "Zoom Out"
* "Zoom To Fit"
* "Use Perspective"

So, make fix SolveSpace for keep selected objects after any view changing.

whitequark, 2016-08-01:
> "Nearest Ortho View"
> "Nearest Isometric View"

I just checked and these keep selection.

> "Align View to Workplane"
> "Center View At Point"

These require a selection to proceed. In fact, if you select anything except a workplane (for the 1st case) or a point (for the 2nd case), it will cause an error. And it is a common pattern in SolveSpace that commands requiring selection to proceed clear that selection afterwards.

So, closing as invalid.

Evil-Spirit, 2016-08-07:
fixed anyway.

whitequark, 2016-08-07:
@Evil-Spirit Oh, good catch, thanks!

Symbian9, 2016-08-07:
Thank You both!

Symbian9, 2016-08-07:
Anyway, "Align View to Workplane" no need selection, because its automatically aligned to active (latest in group list) or nearest workplane

whitequark, 2016-08-13:
Fixed in master and 2.x.

closed 2016-08-13 by whitequark