whitequark/solvespace #228: SolveSpace 2.1 (typo issues)

Symbian9, 2016-07-25:
Here are small issues in UI typography. Please, fix them:
* "... 2d ..." -> "... 2D ..."
* "... 3d ..." -> "... 3D ..."
* "Zoom To Fit" -> "Zoom <u>to</u> Fit"
* "Center View At Point" -> "Center View <u>at</u> Point"
* "Sketch In 3d" -> "Sketch <u>in 3D</u>"
* "Anywhere In 3d" -> "Anywhere <u>in 3D</u>"
* "New group in 3d (Shift+3)" -> "New group in <u>3D</u> (Shift+3)"
* "Lathe" -> "<u>Solid of revolution / Lathe</u>" (if group type is "union" - its "Solid of revolution", if group type "difference" - its "Lathe")
    * https://en.wikipedia.org/wiki/Solid_of_revolution
    * https://en.wikipedia.org/wiki/Lathe
Maybe, there are also other, but now I not remember.

whitequark, 2016-07-25:
Not typos.

closed 2016-07-25 by whitequark

Symbian9, 2016-07-25:
Then, why there are different "in/In": "Sketch In 3d" and "New group in 3d (Shift+3)"?

Anyway, UI must look as software standart, so "2d/3d" must be replaced "2D/3D"

Evil-Spirit, 2016-07-25:
@Symbian9, there is no software standarts.