whitequark/solvespace #229: Help for coding

ryannining, 2016-07-28:
Hi, i have try my best hack code on loadfromfile, so it can change the Rotate_group repeat times parameter by external variable. So far, it can change the valA and keep the expression string . The problem is, when the number repeat times is chaged, we need to call this right (from textscreen.cpp editcontroldone)

```
// If the group does not contain any constraints, then
                    // set the numerical guess to space the copies uniformly
                    // over one rotation. Don't touch the guess if we're
                    // already constrained, because that would break
                    // convergence.
                    if(c == 0) {
                        double copies = (g->skipFirst) ? (ev + 1) : ev;
                        SK.GetParam(g->h.param(3))->val = PI/(2*copies);
                    }
```

So i copy part of that code to loadfromfile, but `sv.g` and `g` in code above is different object. I dont know how to convert `sv.g` to `g` like code above. I have try my best (mostly trial and error, since i dont know much about cpp) still not work. Please help me ...

```
...
// reevaluate expression, who know the "repeat times" change
            if (sv.g.svalA!=""){
                Expr *e=Expr::From(sv.g.svalA.c_str(), false);
                double vv=e->Eval();
                sv.g.valA=vv;
            }
            // scale paremeter
            if (sv.g.sscale!=""){
                Expr *e=Expr::From(sv.g.sscale.c_str(), false);
                double vv=(SS.ExprToMm(e));
                sv.g.scale=vv;
                
            }
             
            // if rotate
            
            // need to reposition the copies from rotate group, because perhaps
            // the number of copies are change (by external parameter/variable)
            double copies = sv.g.valA;
            Param par=SK.GetParam(sv.g.h.param(3));
            par.val = PI/(2*copies); 
            
            
            SK.group.Add(&(sv.g));
...
```

Thank you.
Ryan widi

ryannining, 2016-07-28:
After few trial and error again
i found this compiled OK, but CORE DUMPED when load file...
```
            if (changed) {
                double copies = sv.g.valA;
                Group *g = SK.GetGroup(sv.g.h);
                Param *par=SK.GetParam(sv.g.h.param(3));
                par->val = PI/(2*copies);
            }

```

Now learning the savefile... looks the param that hold the vales PI/2*copies is not loaded yet.. so looks llike it will not work.

Evil-Spirit, 2016-07-28:
@ryannining, probably you must run it after loading and GenerateAll.
But better way is perform saving of this values.

ryannining, 2016-07-28:
ok. I will learn more about the code and try again . thx

ryannining, 2016-07-28:
Ok after more trial and error i store group that valA is changed and add in the end of loadfromfile before schedulegenerateall.
```
    for (int i=0;i<nchanged;i++){
                Group *g=gchanged[i];
                double copies = (g->skipFirst) ? (vchanged[i] + 1) : vchanged[i];
                SK.GetParam(g->h.param(3))->val = PI/(2*copies);
                
    }
   ScheduleGenerateAll();
```

Its works !!  
https://youtu.be/lKpvIbZh47M

closed 2016-07-28 by ryannining