whitequark/solvespace #231: Broken DXF export

Leichtbier, 2016-07-30:
Since SolveSpace 2.1 I'm no longer able to use the exported DXF files with either Inkscape 0.91 or RDWorks (a software for my laser cutter). 

I use Solvespace 2.1~dce90646 with:
File/Export 2d View...
(file type: DXF file (AutoCAD 2007))

Inkscape 0.91 crashes with the following error:
```
Traceback (most recent call last):
  File "dxf_input.py", line 477, in <module>
    layer = layer_nodes[vals[groups['8']][0]]
KeyError: u's001-#def-active-grp'
```

whitequark, 2016-07-30:
This seems like a bug in Inkscape. LibreCAD and AutoCAD open the files just fine...

Leichtbier, 2016-07-31:
Thank you for your fast response.
I think if it's an error or not is open to interpretation. I found a post which describes the problem more precisely: [https://bugs.launchpad.net/inkscape/+bug/1472429]

There is also a solution for the Inkscape import. I'm not familiar enough with the dxf specification to decide if it's a probelm of the SolveSpace export or a problem of the Inkscape import.

Now the Inkscape import works againe. However, this not solves the problem with the laser cutter software.

whitequark, 2016-07-31:
OK. I am not sure if this is strictly a bug but it is abnormal enough to fix.

@Evil-Spirit see https://bugs.launchpad.net/inkscape/+bug/1472429/comments/5 for the description of the problem with the exported files.

Evil-Spirit, 2016-08-07:
fixed

whitequark, 2016-08-13:
@Evil-Spirit Have you verified that these DXF files load in Inkscape? I've tried that, and my Inkscape doesn't actually load anything, but it also doesn't load any files saved in 2.0, so I don't really know.

Evil-Spirit, 2016-08-13:
@whitequark, you have to swith off some checkbox about scaling inside Inkscape import dialog.

whitequark, 2016-08-13:
@Evil-Spirit Makes no difference.

Evil-Spirit, 2016-08-13:
@whitequark, does it shows any error messages of just doing nothing? I switched off "auto scale to a4" and this shows result for me.

whitequark, 2016-08-13:
Just does nothing.

Evil-Spirit, 2016-08-13:
[image]

Evil-Spirit, 2016-08-13:
[image]

whitequark, 2016-08-13:
Okay, so it's just my install of Inkspace broken somehow.

Fixed in master and 2.x.

@Leichtbier I couldn't find an easy way to test your laser cutter software, so please verify that [this file](https://github.com/whitequark/solvespace/files/416417/test.dxf.zip) works with it, and open a new issue if it doesn't. Please also figure out what needs to be adjusted to make it compatible (ask the developers?), or at least provide the exact steps to reproduce the problem you're having.

closed 2016-08-13 by whitequark