whitequark/solvespace #232: Crash in DoProjectedPoint

whitequark, 2016-08-02:
To reproduce:
  * switch to Free in 3D
  * draw line and point
  * add a distance constraint

Broken in https://github.com/whitequark/solvespace/commit/6e17780e01d89878172fe1eb36070361bbcba801

whitequark, 2016-08-02:
I've commented this code for now, so uncomment it when testing.

whitequark, 2016-10-06:
@Evil-Spirit Please look into this.

Evil-Spirit, 2016-10-06:
@whitequark, ok

Evil-Spirit, 2016-10-09:
@whitequark, just performed test. The constraints visualization should be reconsidered in the future to prform drawing all the parts of the same dimension always in one plane.

whitequark, 2016-10-09:
Yeah, there are a few odd cases in visualization that are visually unpelasant.

whitequark, 2016-10-09:
This really isn't any better... should we just disable this?
[image]

whitequark, 2016-10-09:
Actually, as far as I can see, rendering-wise your patch doesn't change *anything*. Is this right?

whitequark, 2016-10-09:
Ok I see the description of the original commit (in 6e17780e0). Actually, what's the problem with drawing it like 
[image]
for the free in 3D case?

Evil-Spirit, 2016-12-19:
@whitequark,
implemented here https://github.com/Evil-Spirit/solvespace-master/commits/issue-wq-232
+some bug fixes

whitequark, 2016-12-21:
Merged

closed 2016-12-21 by whitequark